arduino/Arduino #10219: Arduino IDE 1.8.12 crashes on macOS 10.15.5 when clicking scrollbar

erwinbonsma, 2020-05-16:
Version: Arduino 1.8.12, but also occurs in 1.8.10 and latest hourly build
OS: macOS 10.15.5 (Beta) - Currently available as Developer preview

The IDE starts up okay, and after navigating the various security warnings dialogs from the OS, I can use the ID as usual. However, as soon as I click a scrollbar for one of the text editor tabs, the application crashes with the following error:

```
Application Specific Information:
*** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '-[NSCompositeAppearance _callCoreUIWithBlock:options:]: unrecognized selector sent to instance 0x600001a70880'
abort() called
terminating with uncaught exception of type NSException
```

per1234, 2020-05-20:
@facchinm is this an OpenJDK issue?

The reason I ask is because we have two reports of this on the forum (https://forum.arduino.cc/index.php?topic=682815), but both users say the problem isn't fixed by using Arduino IDE 1.8.10. My understanding was that the switch to OpenJDK happened in Arduino IDE 1.8.11, but perhaps I'm mistaken. I couldn't find where it happened in the commit history, but I don't even know what to look for.

facchinm, 2020-05-22:
@per1234 you are right of course. I added the tag to "condense" the issues in a single place so we'll be able to understand if https://github.com/arduino/Arduino/pull/10234 approach will solve all osx/java issues (which are mostly openjdk related).
Fell free to add another more explicit tag if you wish :wink: