arduino/Arduino #10225: [Library Manager]

MAINAKMONDAL98, 2020-05-16:
Please read the Issues section of the Contributing Rules at the "Contributing" link to the right before submitting an issue report.

MAINAKMONDAL98, 2020-05-16:
https://github.com/MAINAKMONDAL98/MSMPLOTTER

MAINAKMONDAL98, 2020-05-16:
Can you please tell me can I see it on library manager searching its name MSMPLOTTER?
Thanks for labeling.

per1234, 2020-05-17:
@MAINAKMONDAL98 once your library has been added to the Library Manager index, it can indeed be found by searching Library Manager for the library.

Your library will not be available for installation via Library Manager until it has been added to the index. cmaglie occasionally comes through to clear all the outstanding Library Manager inclusion requests and closes those issues when that is done. You will receive a notification from GitHub of this. There may be a delay of some hours after that for the library's releases to be picked up by the indexer and for the updates to the [Library Manager index](http://downloads.arduino.cc/libraries/library_index.json) to propagate through the CDN cache.

MAINAKMONDAL98, 2020-05-17:
thanks @per1234

cmaglie, 2020-05-18:
Done.

closed 2020-05-18 by cmaglie