arduino/Arduino #10228: [IDE] Slow Startup - BaseNoGui.onBoardOrPortChange  trigger two rescanLibraries()

ricardojlrufino, 2020-05-17:
I noticed that the method is being called 2 twice:
librariesIndexer.rescanLibraries ()

[image]

I believe this is a bug, because the `librariesIndexer.setArchitecturePriority`, seems to have no influence on the process.

This makes the other issue (#10214) I found a little worse

matthijskooijman, 2020-05-18:
> I believe this is a bug, because the librariesIndexer.setArchitecturePriority, seems to have no influence on the process.

But isn't this fixed by calling `rescanLibraries` the second time? Maybe that's why it is called at all?

Scanning twice is of course a performance problem to be fixed. Without looking at the code in detail, I would think that maybe `setLibrariesFolders` should just not call `rescanLibraries()`? It seems that the two places that call `setLibrariesFolders` now also call `rescalLibraries`, so just removing the first call would be ok?

cmaglie, 2020-05-18:
After looking at the call tree of `setLibrariesFolder`, the following places `setLibrariesFolder` is followed by `rescanLibraries`:

https://github.com/arduino/Arduino/blob/02a2e22649feb4d85d807b51122f2501b06da866/app/src/processing/app/Base.java#L365-L366

https://github.com/arduino/Arduino/blob/02a2e22649feb4d85d807b51122f2501b06da866/arduino-core/src/processing/app/BaseNoGui.java#L674-L681

https://github.com/arduino/Arduino/blob/02a2e22649feb4d85d807b51122f2501b06da866/app/test/cc/arduino/contributions/UpdatableLibraryTest.java#L39-L40

https://github.com/arduino/Arduino/blob/02a2e22649feb4d85d807b51122f2501b06da866/app/test/cc/arduino/contributions/UpdatableLibraryTest.java#L66-L67

---

Instead in the following places the "autoreload" feature is actually used:

https://github.com/arduino/Arduino/blob/02a2e22649feb4d85d807b51122f2501b06da866/app/test/cc/arduino/contributions/UpdatableLibraryTest.java#L48-L49

https://github.com/arduino/Arduino/blob/02a2e22649feb4d85d807b51122f2501b06da866/app/test/cc/arduino/contributions/UpdatableLibraryTest.java#L75-L76

---

@ricardojlrufino 
if you add a call to `rescanLibraries` in the two cases above, I think you can safely remove the `rescanLibraries` call from `setLibrariesFolder`. Nice catch, if you want to open a PR this can be quickly merged.

matthijskooijman, 2020-05-18:
> Instead in the following places the "autoreload" feature is actually used:

Ah, I had not looked at the tests in my analysis above, good call. But indeed, just calling `rescalLibraries` in the tests seems fine :-)

ricardojlrufino, 2020-05-18:
to have something cleaner, what about adding a second boolean parameter to `setLibrariesFolders`
like : indexer.setLibrariesFolders(File files, boolean rescan); 
what do you think ?

ricardojlrufino, 2020-05-18:
[image]

ricardojlrufino, 2020-05-18:
[image]

matthijskooijman, 2020-05-19:
That would also be ok (though boolean parameters have the risk of becoming unclear if you don't add a comment in the call), but I'm not so sure it would be needed if `rescanLibraries` is called always anyway (ignoring the tests). OTOH, in one of the two places in the code, `rescanLibraries` is called *directly* after `setLibrariesFolders`, so that could be shortened to pass true instead).

Still, I'm not sure. On one hand, it might be good to somewhat ensure that rescan happens when changing the library folder. OTOH, explicitly calling `rescanLibraries()` might result in code that is easier to read (clearer what is going on).

One other alternative would be to have a `setLibrariesFolderAndRescan()` function, which would be easier to read that a `true` argument.

cmaglie, 2020-05-19:
I agree, we already have made some API in the IDE in the past using boolean that lead to methods like `doSomething(true, false, false)`. That looks very bad in retrospecitve.

@ricardojlrufino I'd follow Matthijs suggestion and do:

* add `rescanLibraries()` where it's missing and remove it from `setLibrariesFolder()`

or

* add `setLibrariesFolderAndRescan()` and remove rescan from `setLibrariesFolder()`

up to you, both looks OK to me.

PR #10270 (2020-05-25): Fix double libraries rescan.
I've cherry-picked @ricardojlrufino commits to fix #10228 and:

* added a trivial fix to function call
* fixed indentation
commit b914070c (2020-05-25): Avoid usueless double calls to rescanLibraries()

Previously rescanLibraries() was automatically called internally in
setLibrariesFolder(). This lead to double calls to rescanLibraries()
when setLibrariesFolder() was used in combination with an explicit
call to rescanLibraries().

This commit adds a new method setLibrariesFoldersAndRescan(..) and
removes the internal call to rescanLibraries() from setLibrariesFolder().

The existing setLibrariesFolder()+rescanLibraries() combos have been
replaced with setLibrariesFoldersAndRescan().

Fix #10228
closed 2020-05-25 by cmaglie