arduino/Arduino #10236: [IDE] Header Tab is covered by "menu icon".

ricardojlrufino, 2020-05-18:
See:
[image]

matthijskooijman, 2020-05-18:
What would be a good way to solve this? Stop drawing tabs if they do not fit before the "menu icon"? Might be good to also draw some indication that there are more tabs hidden. Can you access these right now at all? Through "menu icon" I think? Maybe also by scrolling the tab bar (probably not, since we're drawing that manually and probably didn't include scrolling in there)?

facchinm, 2020-05-18:
https://github.com/arduino/Arduino/commit/578d34e2031b20e9371345436a72f9482ed1e9a1 and following patchset (merged in IDE beta) solves the issue, I'd love to finally merge it (but testers are needed of course).
Relevant patches:

[Limit selected tab size in case of crowded header
](https://github.com/arduino/Arduino/commit/cabd626f0a88a3a179825135a403cc16d2af4930)[tabs] [some heavy lifting on tabs handling …](https://github.com/arduino/Arduino/commit/336cde194dbeab036b4e7bf9110744766aadb13b)
[Implement MouseWheelListener on tabs …](https://github.com/arduino/Arduino/commit/56dcf6bde430a735b4786af593c0e6c90c6df40e)
[Shift tabs if selected one if being painted offscreen …
](https://github.com/arduino/Arduino/commit/578d34e2031b20e9371345436a72f9482ed1e9a1)

ricardojlrufino, 2020-05-18:
It seems that tabs have many requirements that create problems. One option would be:
JIDE, has some components of advanced tabs ...

[image]

Some features
- Create a menu when you have no more space
- Add components at the beginning and at the end
- Actions to close the tab
- Scroll support.

Example

https://gist.github.com/ricardojlrufino/b65c41219ec11e429694e6daab36ee4e

[image]

Need create customUI:
https://github.com/jidesoft/jide-oss/blob/4fc39cb4b830a961ed713576fbb89e2726788837/src/com/jidesoft/plaf/eclipse/EclipseJideTabbedPaneUI.java

ricardojlrufino, 2020-05-18:
https://www.jidesoft.com/products/oss.htm

ricardojlrufino, 2020-05-19:
Updated examples:
https://gist.github.com/ricardojlrufino/b65c41219ec11e429694e6daab36ee4e

I was able to implement a customUI for the component. Now we have all paintX, which can be used to replicate that of the Arduino.

One of the things I managed to do was set the TabbedPane content to 0px, so we don't have to worry about the content of the tabs.

PS: It was just a proof of concept, I'm not focused on this problem

[image]

@facchinm Will it meet the requirements?

facchinm, 2020-05-22:
@ricardojlrufino did you take a look at the other implementation I linked? I'd be willing to merge that since it seems to resolve all the problems at once (at the cost of some added complexity)