arduino/Arduino #10242: Warning: platform.txt from core 'MediaTek ARM7 EJ-S (32-bits) Boards' contains deprecated recipe.ar.pattern="

bryn51, 2020-05-19:
Hi All,

I am just trying out a LinkIt One board supplied by MediaTek/Seeed Studios.  It contains a Cortes M7 Core.

When compiling the example Sketches (randomly selected) I get the following errors:
Warning: platform.txt from core 'MediaTek ARM7 EJ-S (32-bits) Boards' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.

I wpould consider upgrading the core, except I already followed the instructions on MediaTek "Getting Started with LinkIt One" and assume I have the latest.

The compiler issues a "Warning" type error, however at the end, it gives up the following:
C:\Program Files (x86)\Arduino_1.8.9>"C:\Users\Bryn\AppData\Local\Arduino15\packages\LinkIt\tools\linkit_tools\1.1.23\PushTool.exe" -d arduino -c -p "C:\Users\Bryn\AppData\Local\Temp\arduino_build_632412/Battery.ino.vxp" 

I also note that the Tx/Rx LEDS do not flash whilst uploading the sketch.
However I also note that I appear to be using the correct type of MicroUSB cable, because two serial ports are recognised.  Althopugh I select Port 8 (Modem), the uploader appears to search through available ports before selecting Port 7:
PORTS {COM6, COM7, COM8, } / {COM6, COM7, COM8, } => {} <=== (Multiple lines of this )
Uploading using selected port: COM7

I assume that the sketch is not actually being uploaded, therefore I am unable to test the board.

There is the possibility it may be bricked, because in an earlier part of the setup procedure the user is encouraged to upload a new firmware, which I did, and which came up successful.

Some help on the several issues would be appreciated.

per1234, 2020-05-19:
This issue tracker is only to be used to report bugs or feature requests. The LinkIt board and its support platform are 3rd party products, so Arduino can't provide support for problems specific to those products. This topic is more appropriate for the Arduino Forum. I'm sure we'll be able to help you with your problem over there.

Please do this:
1. Read http://forum.arduino.cc/index.php?topic=148850
2. If you haven't already done so, create a new thread in the appropriate section of the forum (http://forum.arduino.cc/), following all the rules.

closed 2020-05-19 by per1234

bryn51, 2020-05-19:
Thanks for your Terse reply, However I followed the advice given in Paragraph 4 and found a nearly identical problem in this same thread.

Sam00Cleveland, 2020-12-12:
did anybody find a solution to this ?

having the same issue trying to use the board, purchased it about 5 years ago and never actually been able to upload anything to it, always some form of compiler issue

OSX Catalina with the latest Arduino IDE and most recent boards definition (downloaded through boards manager)