arduino/Arduino #10245: feature request: servo class: servo.write(angle, rotationspeed)

dsyleixa, 2020-05-19:
feature request: servo class: servo.write(int angle, float rotationspeed)
angle: degrees (endpoint)
rotationspeed: degrees per second
e.g.,
servo.write(90); // let the servo move to postion 90° immediately
servo.write(180, 10.0); // moves the servo from the current position (90°) to  180° by a rotation speed of 10°/sec

ernesto-tapia, 2020-06-03:
Hello, 
I would like to work on this feature request, is it still available?

dsyleixa, 2020-06-03:
yes, just make a pull request (PR)!

ernesto-tapia, 2020-06-22:
This was a little bit too much for my current level, and Im not sure that non blocking can be achieved without the use of another update function running in the code. @per1234, do you have any insight on how I can possibly attack this issue?

dsyleixa, 2020-11-13:
no one to achieve this...?

dsyleixa, 2021-01-24:
I'll close it her and move it to github Servo lib issue tracker

closed 2021-01-24 by dsyleixa