arduino/Arduino #10246: Please help! I am having this issue when i am interfacing ssd1306 with arduino

ghulamhyderpalli, 2020-05-19:
I am having this issue when I verify my code in Arduino IDE

In file included from C:\Users\Ghulam Hyder Palli\Documents\Arduino\libraries\Adafruit_GFX_Library\Adafruit_MonoOLED.cpp:20:0:

C:\Users\Ghulam Hyder Palli\Documents\Arduino\libraries\Adafruit_GFX_Library\Adafruit_MonoOLED.h:30:32: fatal error: Adafruit_I2CDevice.h: No such file or directory

compilation terminated.

exit status 1
Error compiling for board Arduino Pro or Pro Mini.

ladyada, 2020-05-19:
You will also need to install the latest Adafruit BusIO library. Search for "Adafruit BusIO" in the library manager, or install by hand from https://github.com/adafruit/Adafruit_BusIO

per1234, 2020-05-19:
Thanks for the assist @ladyada!

Since I think this issue is resolved, and it doesn't represent a bug report or feature request for Arduino's software, I'll go ahead and close this issue.

closed 2020-05-19 by per1234

ghulamhyderpalli, 2020-05-19:
Thank you so much

marcussporring, 2020-05-27:
Thanks! +1