arduino/Arduino #10248: Multiple Commands in Recipes Fail

oclyke, 2020-05-20:
Hello - I am making this report both to
a) confirm whether or not this behaviour is expected
b) respectfully suggest a feature enhancement to allow this sort of action

It may be useful for [recipes](https://arduino.github.io/arduino-cli/platform-specification/#build-process) to support multiple commands, e.g. ```executable1.exe & arm-none-eabi-g++``` (windows) or ```executable1 ; arm-none-eabi-g++``` (linux / mac)

As it stands there seems to be some mechanism blocking this from happening. I have reproduced the issue in a stripped-back branch of my core [Arduino_Apollo3:test-multiple-command-recipes](https://github.com/sparkfun/Arduino_Apollo3/tree/test-multiple-command-recipes)
* Install this repo as a core package and change to the proper branch 
	* easiest to install w/ board manager and then replace contents w/ git
	* addtl. url: https://raw.githubusercontent.com/sparkfun/Arduino_Boards/master/IDE_Board_Manager/package_sparkfun_index.json
	* search: apollo3
* Open a new sketch
* Select "SparkFun RedBoard Artemis ATP" from the Tools menu
* Press "Verify"
* Observe error ```test.exe: error: unrecognized arguments:```
* Copy the command that caused the error from the Arduino window
* Open a command line interface on your machine (Cmd.exe, Terminal, Bash etc)
* Paste the command + run
* Observe proper execution of multiple commands (error from cross compiler is unimportant)

I am unfamiliar with the internals of the IDE. Stumped b/c the command generated by the IDE appears to work correctly when executed directly by the host system.

This is related to another possible feature request that would solve the same issue -- that is to provide [hooks](https://arduino.github.io/arduino-cli/platform-specification/#pre-and-post-build-hooks-since-arduino-ide-165) at more stages of the build process. Currently the earliest hook that one can access is ```recipe.hooks.sketch.prebuild.NUMBER.pattern``` which occurs after the library detection and function prototype generation steps. However at least in my use case these steps rely on generated files - a hook that occurs right after a user presses "verify" would solve my issue.

Of course I need to find a workaround using existing versions of the IDE - for coverage.

Thanks for listening.

matthijskooijman, 2020-05-21:
This is indeed not supported. I think that a recipe is parsed into a single command that is executed directly, rather than passed as-is to a shell (since that latter would open a can of worms in terms of shell special characters, shell injection, etc).

Splitting commands on `;` could be implemented, but I'm not sure that it needs to be. In practice, cores that need this would ship a shell script (typically both a shell script and a batch file). For example of such a thing (for upload, but for compile would be similar) see https://github.com/stm32duino/Arduino_Core_STM32/blob/00236f4320d8e9e17e364f51cfe58fb236157271/platform.txt#L156-L157 and https://github.com/stm32duino/Arduino_Tools/blob/master/linux/massStorageCopy.sh and https://github.com/stm32duino/Arduino_Tools/blob/master/win/massStorageCopy.bat (this has the scripts in a separate tools repo, but you could also just put them in the same repo somewhere).

oclyke, 2020-05-21:
Okay, thanks for the confirmation.

I'm familiar with using scripts (particularly Python for ease of generating executables for several systems) within the core. In my particular case this time the script would need to run before the preprocessing steps but there do not appear to be any hooks available that early in the build process. 

@matthijskooijman do you know of any way to run a script as part of the core installation? Or some other way that I could generate a list of absolute include paths that references the platform install location? 

I am hoping to avoid listing 300+ include paths directly in platform.txt (where I can access {runtime.platform.path} for compatibility)

matthijskooijman, 2020-05-31:
> @matthijskooijman do you know of any way to run a script as part of the core installation? Or some other way that I could generate a list of absolute include paths that references the platform install location?

Nothing springs to mind, but I think you've also reported another issue about this and found some other way around this, right? Also, it seems all of this is rather unrelated to the issue subject of running multiple commands in a single recipe.

Given that this is somewhat intentionally *not* supported and easy to work around using scripts if needed, I think this issue can be closed? Or is there anything left to implement or discuss?

oclyke, 2020-05-31:
Well I have found a solution in my particular case but I do believe there is a feature add that is worth discussing here:

- Provide a recipe hook *before* any other steps. 

As part of my hunt for a solution I noticed that the earliest hook that I could use occurred after the library detection and preprocessing steps. You suggested that there are easy ways to work around this using scripts... but that's not true when there is no way to call those scripts in time. 

It is possible that others would find a hook that occurs right away after the user presses verify / upload to be useful in their own applications.

Thanks for listening!

oclyke, 2020-05-31:
I was about to change the title of this issue to reflect the new direction that it has taken... Instead how about we close this issue and I open one on the arduino-cli repo requesting this particular hook?

matthijskooijman, 2020-05-31:
> I was about to change the title of this issue to reflect the new direction that it has taken... Instead how about we close this issue and I open one on the arduino-cli repo requesting this particular hook?

Yes, that would better I think. Thanks for following up!

closed 2020-05-31 by matthijskooijman