arduino/Arduino #10250: [Library manager] Please add Anyleaf

David-OConnor, 2020-05-21:
Github repo: https://github.com/AnyLeaf/ph-cpp

Please let me know if there's anything you'd like me to change, or if you need more info.

cmaglie, 2020-05-22:
Done.

closed 2020-05-22 by cmaglie

njh, 2020-08-10:
It looks like the AnyLeaf `ph-cpp` repo for Arduino has now been deleted.

Is there any information about why?

David-OConnor, 2020-08-10:
I restructured the way its repo interacts with it's sister repos, and had a
typo in the name. Fixed.

njh, 2020-08-10:
Ah, ok.

I am still getting a 404 for:
https://github.com/AnyLeaf/ph-cpp

David-OConnor, 2020-08-10:
Oops, was set in Private mode! Fixed.

(If you'd like more details, I have drivers for this hardware in 3 languages, all which share a same repo to make it easier to manage. I then push the language-specific subfolders to individual repos, one of which is this.)