arduino/Arduino #10251: How to install a board with the zip file

vivekvardhanadepu, 2020-05-21:
[image]
If I copy this zip file to another computer and place in the corresponding folder, will arduino automatically detect it and install it for me? If so, will this also work for zip file downloaded from source repo?

vivekvardhanadepu, 2020-05-21:
[image]
Also, does this method work?

per1234, 2020-05-21:
>If I copy any of these zip files to another computer and place in the corresponding folder, will arduino automatically detect it and install it for me?

The Arduino IDE will use the files from the staging folder if present, rather than downloading them from the Internet. However, I'm not sure what you mean by "automatically detect it and install it for me". You still need to go through the standard Boards Manager installation process.

>will this also work for zip file downloaded from source repo?

Yes. You can find the download URLs for the package files in the JSON file at the package's Boards Manager URL.

>does this method work?

Yes.

---
I'll go ahead and close this now. If you need further assistance, please ask on the [Arduino Forum](http://forum.arduino.cc/index.php?topic=148850). This issue tracker is intended to be used only for reporting bugs and making feature requests for the official Arduino software and documentation. The forum is the appropriate place to request support. I'm sure we'll be able to help you with any further issues you might have over there.

closed 2020-05-21 by per1234