arduino/Arduino #10274: [Library Manager] update URL for Plotter library

devinaconley, 2020-05-26:
Updated `Plotter` library to 1.5+ format, which now enables using a single repo for this project. New repo here:
https://github.com/devinaconley/arduino-plotter

For reference, original issue:
https://github.com/arduino/Arduino/issues/4481

per1234, 2020-05-26:
Hi @devinaconley. Although your library is now in 1.5 format, at the time of the last tag/release, it didn't have a library.properties file in the root of the repository, as required. Please either create a new tag/release or else redo the existing one.

devinaconley, 2020-05-26:
Hi @per1234 - thanks for the feedback, just tagged a new release

https://github.com/devinaconley/arduino-plotter/releases/tag/2.4.0

cmaglie, 2020-05-31:
Done.

closed 2020-05-31 by cmaglie

devinaconley, 2020-06-01:
Thanks @cmaglie @per1234