arduino/Arduino #10275: [Library Manager] Please add Arduino_TensorFlowLite_ESP32 library

tanakamasayuki, 2020-05-26:
Hello,
please add our library into your library manager.

Link to the repository:
https://github.com/tanakamasayuki/Arduino_TensorFlowLite_ESP32

And to library.properties file:
https://github.com/tanakamasayuki/Arduino_TensorFlowLite_ESP32/blob/master/library.properties

Thanks!

per1234, 2020-05-26:
Hi @tanakamasayuki. Arduino started a policy some time ago that 3rd party libraries with a library.properties `name` value starting with the word "Arduino" (or any capitalization thereof) will no longer be accepted for inclusion in the Arduino Library manager index. The reason is that Arduino wants to reserve these names for their own official libraries.

If you want your library to be added to the Library Manager index, you will need to change the `name` value in library properties to one that is compliant with the Arduino Library Specification:
https://arduino.github.io/arduino-cli/library-specification/#libraryproperties-file-format

Then update the `version` value in library.properties and create a new release/tag, or else redo the existing release/tag.

Note that this restriction only applies to the `name` value. You are not required to rename the repository or header files.

Once you have done this, please comment here so we'll know we can proceed.

More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

tanakamasayuki, 2020-05-26:
sorry.
We changed the name and re-released it.

cmaglie, 2020-05-31:
Done.

closed 2020-05-31 by cmaglie