arduino/Arduino #10279: Arduino MKR FOX 1200 seen as MKR WAN 1300

jacopisky, 2020-05-27:
Bought a new arduino MKR FOX 1200 and arduino IDE seems to recognize it as MKR WAN 1300; not able to load any sketch

jacopisky, 2020-05-27:
Checked and the board declares
BN: Arduino MKR WAN 1300
VID: 0x2341
PID: 0x8053
so according to boards.txt it is a mkr wan but the label on the ORIGINAL board says fox. How can I load sketch in it overriding arduino ide

per1234, 2020-05-27:
Please do this:
1. **File > Preferences**
1. Uncheck the box next to **Show verbose output during: > compilation**
1. Check the box next to **Show verbose output during: > upload**
1. Click the "OK" button.
1. **Sketch > Upload**
1. After the upload fails you'll see a button on the right side of the orange bar "Copy error messages". Click that button.
1. [Paste](https://lifehacker.com/how-to-copy-cut-and-paste-for-beginners-5801525) the error messages in a reply here.

jacopisky, 2020-05-27:
Arduino:1.8.11 (Windows 10), Scheda:"Arduino MKR FOX 1200"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Utente\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Utente\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Utente\Documents\Arduino\libraries -fqbn=arduino:samd:mkrfox1200 -vid-pid=0X2341_0X8053 -ide-version=10811 -build-path C:\Users\Utente\AppData\Local\Temp\arduino_build_324504 -warnings=none -build-cache C:\Users\Utente\AppData\Local\Temp\arduino_cache_811195 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Users\Utente\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.2.1 -prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Users\Utente\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.2.1 -prefs=runtime.tools.openocd.path=C:\Users\Utente\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino7 -prefs=runtime.tools.openocd-0.10.0-arduino7.path=C:\Users\Utente\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino7 -prefs=runtime.tools.CMSIS-Atmel.path=C:\Users\Utente\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.2.0 -prefs=runtime.tools.CMSIS-Atmel-1.2.0.path=C:\Users\Utente\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.2.0 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\Utente\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=C:\Users\Utente\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.bossac.path=C:\Users\Utente\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.7.0-arduino3 -prefs=runtime.tools.bossac-1.7.0-arduino3.path=C:\Users\Utente\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.7.0-arduino3 -prefs=runtime.tools.CMSIS.path=C:\Users\Utente\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.CMSIS-4.5.0.path=C:\Users\Utente\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0 -verbose C:\Users\Utente\Documents\Arduino\libraries\Arduino_SigFox_for_MKRFox1200\examples\FirstConfiguration\FirstConfiguration.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Utente\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Utente\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Utente\Documents\Arduino\libraries -fqbn=arduino:samd:mkrfox1200 -vid-pid=0X2341_0X8053 -ide-version=10811 -build-path C:\Users\Utente\AppData\Local\Temp\arduino_build_324504 -warnings=none -build-cache C:\Users\Utente\AppData\Local\Temp\arduino_cache_811195 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Users\Utente\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.2.1 -prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Users\Utente\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.2.1 -prefs=runtime.tools.openocd.path=C:\Users\Utente\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino7 -prefs=runtime.tools.openocd-0.10.0-arduino7.path=C:\Users\Utente\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino7 -prefs=runtime.tools.CMSIS-Atmel.path=C:\Users\Utente\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.2.0 -prefs=runtime.tools.CMSIS-Atmel-1.2.0.path=C:\Users\Utente\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.2.0 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\Utente\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=C:\Users\Utente\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.bossac.path=C:\Users\Utente\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.7.0-arduino3 -prefs=runtime.tools.bossac-1.7.0-arduino3.path=C:\Users\Utente\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.7.0-arduino3 -prefs=runtime.tools.CMSIS.path=C:\Users\Utente\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.CMSIS-4.5.0.path=C:\Users\Utente\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0 -verbose C:\Users\Utente\Documents\Arduino\libraries\Arduino_SigFox_for_MKRFox1200\examples\FirstConfiguration\FirstConfiguration.ino
Using board 'mkrfox1200' from platform in folder: C:\Users\Utente\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.6
Using core 'arduino' from platform in folder: C:\Users\Utente\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.6
Detecting libraries used...
"C:\\Users\\Utente\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10811 -DARDUINO_SAMD_MKRFox1200 -DARDUINO_ARCH_SAMD -DUSE_ARDUINO_MKR_PIN_LAYOUT -D__SAMD21G18A__ -DUSB_VID=0x [... cut ...]
"C:\\Users\\Utente\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-size" -A "C:\\Users\\Utente\\AppData\\Local\\Temp\\arduino_build_324504/FirstConfiguration.ino.elf"
Lo sketch usa 44908 byte (17%) dello spazio disponibile per i programmi. Il massimo è 262144 byte.
Le variabili globali usano 3324 byte di memoria dinamica.
No device found on COM9
Errore durante il caricamento dello sketch

I have to reset the sketch otherwise the loading will never end

jacopisky, 2020-05-27:
Arduino:1.8.11 (Windows 10), Scheda:"Arduino MKR FOX 1200"

Lo sketch usa 44908 byte (17%) dello spazio disponibile per i programmi. Il massimo è 262144 byte.
Le variabili globali usano 3324 byte di memoria dinamica.
Forzo il reset aprendo e chiudendo a 1200bps la porta COM9
PORTS {COM9, } / {COM9, } => {}
PORTS {COM9, } / {COM9, } => {}
PORTS {COM9, } / {COM9, } => {}
PORTS {COM9, } / {COM9, } => {}
PORTS {COM9, } / {COM9, } => {}
PORTS {COM9, } / {COM9, } => {}
PORTS {COM9, } / {COM9, } => {}
PORTS {COM9, } / {COM9, } => {}
PORTS {COM9, } / {COM9, } => {}
PORTS {COM9, } / {COM9, } => {}
PORTS {COM9, } / {COM9, } => {}
PORTS {COM9, } / {COM9, } => {}
PORTS {COM9, } / {COM9, } => {}
PORTS {COM9, } / {COM9, } => {}
PORTS {COM9, } / {COM9, } => {}
PORTS {COM9, } / {COM9, } => {}
PORTS {COM9, } / {COM9, } => {}
PORTS {COM9, } / {COM9, } => {}
PORTS {COM9, } / {COM9, } => {}
PORTS {COM9, } / {COM9, } => {}
Uploading using selected port: COM9
C:\Users\Utente\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.7.0-arduino3/bossac.exe -i -d --port=COM9 -U true -i -e -w -v C:\Users\Utente\AppData\Local\Temp\arduino_build_324504/FirstConfiguration.ino.bin -R 
No device found on COM9
Errore durante il caricamento dello sketch

Questo report potrebbe essere più ricco di informazioni abilitando l'opzione
"Mostra un output dettagliato durante la compilazione"
in "File -> Impostazioni"

jacopisky, 2020-05-27:
I have to reset the board on uploading otherwise it will take forever

jacopisky, 2020-05-27:
Just to be clear
[image]
[image]

per1234, 2020-05-27:
Try this:
1. Press and release the reset button on your MKR Fox 1200 twice quickly. You should now see the "L" LED pulsing to indicate the bootloader is running.
1. Select the board's port from the Arduino IDE's **Tools > Port** menu. The port changes when the bootloader is running, so don't assume you already have the correct port selected.
1. **Sketch > Upload**

The upload should now finish successfully. After that, the board should be identified correctly as a MKR Fox 1200 and you should be able to upload normally.

jacopisky, 2020-05-27:
You solved this; what was that ?

per1234, 2020-05-27:
>You solved this

I'm glad to hear it!

>what was that ?

The way uploads work is:

1. The Arduino IDE (or the upload tool used by the Arduino IDE) sends a signal for the microcontroller to reset.
2. The microcontroller resets.
3. The bootloader runs and waits (normally only for a short time) for an upload to start.
4. The upload tool sends the program to the microcontroller, where the bootloader writes it to flash memory.

On the boards with native USB like your MKR Fox 1200, the way the IDE signals the microcontroller to reset is by opening a serial connection at 1200 baud. There is some special code running in the background on the microcontroller that does the reset when it detects that 1200 baud connection.

In some cases, the reset code that's supposed to be on the microcontroller could be missing or not working correctly, which will mean the bootloader never runs and the upload fails with the error you're seeing. The tricky thing about the native USB boards is that, unlike the classic Arduinos (e.g., Uno, Mega), the USB code necessary for the upload to work is running on the same microcontroller as your sketch. So your sketch code can break uploads. The fix is to manually activate the bootloader by using the double reset trick.

Now that you know how to recover your board from this condition, it will be no trouble if you later upload some problematic code and have a similar problem later. Another variant is where you kill the USB CDC code that creates the virtual COM port completely. That has the same fix.

Enjoy! Per

jacopisky, 2020-05-27:
Thank you very much

closed 2020-05-27 by jacopisky