arduino/Arduino #10281: Can't Find JArduino in File -> Examples

8bitallstar, 2020-05-27:
Hi, according to the 2 minute tutorial there should be a JArduino option in File -> Examples -> JArduino -> JArduino firmware, but JArduino doesn't seem to be there, and the  JArduino folder that it mentions doesn't exist either, so if anyone can explain why I don't have these files it would be a big weight off my chest.

Links:
https://github.com/SINTEF-9012/JArduino/wiki/2-Minute-Tutorial

matthijskooijman, 2020-05-27:
This is probably not the right place for getting support for that library, maybe you should ask in that repo instead. Regardless, two things to check:
 - Did you install the library at all, or in the right place? See also https://www.arduino.cc/en/guide/libraries
 - Did you restart the Arduino IDE after installing the library?

closed 2020-05-27 by per1234

per1234, 2020-05-27:
I am providing @8bitallstar with support at their Arduino forum topic:
https://forum.arduino.cc/index.php?topic=686766

Since the request is off topic for this repository, I've gone ahead and closed it. We will use the forum for any further discussion.