arduino/Arduino #10286: [Library Manager]

tfeldmann, 2020-05-28:
Hi, I'd like to submit a library to the library manager:

Timeout
https://github.com/tfeldmann/Arduino-Timeout

Thanks!

per1234, 2020-05-28:
Hi @tfeldmann. Each library added to the Library Manager index must have a unique library.properties `name` value (regardless of case). The `name` value "Timeout" used in your library has already been taken:
https://github.com/NitrofMtl/TimeOut

If you would still like your library added to the index, you'll need to change the `name` value to a unique name, then either update the `version` value in library.properties and make a new release/tag or else redo the existing tag.

After that is done, please comment here to let us know we can proceed. Thanks!

tfeldmann, 2020-05-29:
Thank you for the fast reply. I renamed the library to "Minimal Timeout" and bumped the version 👍

cmaglie, 2020-05-31:
Done.

closed 2020-05-31 by cmaglie