arduino/Arduino #10287: Marlin tree builds/runs on Win10, build hangs indefinitely on same file in MacOS and Linux

allartburns, 2020-05-28:
Using Arduino 1.8.12 on MacOS 10.12.6, Ubuntu 18.04.4 LTS, and Windows 10 version 1909 (2).

This repo https://github.com/allartburns/Marlin.git using branch "bigbox" compiles under Windows 10 and the binary operates as expected on my E3D BigBox printer with a RUMBA controller.

Under MacOS and Linux, the build hangs on the file src/module/temperature.cpp.  By "hang", I mean I can see cc1 using %100 CPU and not finishing within 10 minutes.  If I exit Arduino on MacOS the cc1plus binary does not exit.  Right now it's been running for ~6 minutes and is using 16G+ memory (according to top(1)).

If I revert to the previous change, f66b404, everything compiles and runs as expected on all platforms.

allartburns, 2020-05-31:
Made a simple change today to Configuration.h and now the Win10 build locks up.  Undid the change to what was working two days ago, still locks up.  In the same file as MacOS and Linux, temperature.cpp.

What's infuriating is that this exact same tree compiled just fine two days ago.

I'm wondering if there's a caching problem.

Going to build Arduino from source on Ubuntu 18 and see what I can figure out.

mujido, 2020-06-02:
Try to make sure you don't have a recursive include loop. I've seen this behavior before when two include files refer to each other and don't have a proper include guard.

allartburns, 2020-06-06:
Ah, good point.  I suspect there aren't many dual extruder RUMBAs out there and I might be using a configuration that breaks the pre-processing instructions.

allartburns, 2020-07-05:
I think this is just bad cpp code that recurses forever, I've filed a bug against Marlin:  [Marlin bug](https://github.com/MarlinFirmware/Marlin/issues/18556)

allartburns, 2020-07-08:
Compiling now, still having problems with fan control.  Closing this.

closed 2020-07-08 by allartburns