arduino/Arduino #10291: Support class enum's

paulelong, 2020-05-30:
I'm developing software that works across multiple devices and I would like to share headers which contain enum classes.  However this isn't supported, and while it doesn't stop it from compiling, I'm unable to reference the enum class values in Arduino.  So I'm asking for support of enum classes in Arduino projects.

matthijskooijman, 2020-05-31:
I believe enum classes are actually supported, at least in the AVR core which uses C++11 (and I think also the SAMD core).

It's quite unclear what you're saying exac tly. What does "unable to reference" mean exactly? Also, if it does compile, but not work as expected at runtime, there might very well be a problem in your code, rather than missing support for class enum.

So: Please submit more detailed information, such as:
 - Version of the IDE used
 - Selected board and associated core version.
 - A short example sketch that shows the problem
 - The full build log, including any errors, showing the problem.

paulelong, 2020-05-31:
I apologize, it does work.  My first attempt resulted in an error and when I did a search a came across this post, https://forum.arduino.cc/index.php?topic=260406.0.  Granted it's old but then I could find a release note saying enum class support was added, so I assume it was just a known missing feature.  But now I see I just made a casting error.  Sorry for the churn, but happy about the fast response.

closed 2020-05-31 by paulelong