arduino/Arduino #10299: [Library Manager] Library add request

gin66, 2020-06-01:
Please add my library to the list, if ok:

    https://github.com/gin66/FastAccelStepper

cmaglie, 2020-06-08:
Done.

closed 2020-06-08 by cmaglie

gin66, 2020-06-09:
Thanks. The library is now visible. But it only offers to install the 0.1.x versions. Latest is 0.3.9

How can I do ?

cmaglie, 2020-06-09:
I see, the issue is that from a certain version the library contains symlinks:

```
JOB 3079 - 2020/06/09 09:33:17 path=github.com/gin66, archivename=FastAccelStepper-0.3.9
JOB 3079 - 2020/06/09 09:33:17 Error syncing library: Error while zipping library: Symlink not allowed: /home/jenkins/jenkins/jobs/Libraries/workspace/gitclones/github.com/gin66/FastAccelStepper/tests/FastAccelStepper.cpp -> ../src/FastAccelStepper.cpp
```

we do not allow symlinking in library releases, you must find a way to avoid them and publish another release (or update an existing one).

gin66, 2020-06-09:
Thanks a lot for this clarification. I will update accordingly