arduino/Arduino #10305: （BUG for IDE）The Mega 2560 is having problems with two PWM signals.

cbirdfly, 2020-06-03:
/*************************************************************************/
Description of the problem.
         When compiling the following code for Mega 2560 again with this version of the IDE, the execution results are problematic. When the following code is compiled for the UNO, the execution results are as expected. When the following code was compiled for Mega 2560 using the 1.8.6 version of the IDE, the results were also as expected. So I suspect that there is a bug in that version of the IDE, but I don't know exactly why.
问题描述：
         在用该版本 IDE 给 Mega 2560编译下列代码时，执行结果出现了问题。而给 UNO 编译下列代码时，执行结果与预期一致。再使用1.8.6版本IDE 给 Mega 2560编译下列代码时，执行结果也和预期一致。所以我怀疑是该版本的IDE存在BUG，但是我不知道具体的原因。
/**************************************************************************/
code function：The following program is used to control the operation of two motors, the drive chip is L298N, the correct implementation should be that both wheels can be turned and analogWrite(pin,value) can control the rotation speed of both wheels.
The fault execution is that only one wheel is spinning and the other wheel is stationary.
代码作用：下面的程序时用来控制两个电机工作的，驱动芯片是L298N。正确的执行情况应该是：两个轮子都能够转动，并且analogWrite(pin,value)可以控制两个轮子的转动速度。
故障执行情况是：只有一个轮子在转动，另一个轮子静止。

`#include <Arduino.h>

void Motor_Init();
void Goforward();
void Goback();
void Stopcar();
void TurnL_In_Situ();
void TurnR_In_Situ();

//左电机
#define MotorLpin1   8 //控制位3
#define MotorLpin2   9 //控制位4
#define MotorLpwm    3  //使能调速 ENB
//#define MotorLcountA 21 //编码器A 中断号：5
//#define MotorLcountB 20 //编码器B 中断号：4

//右电机
#define MotorRpin1   10 //控制位1
#define MotorRpin2   11 //控制位2
#define MotorRpwm    6 //使能调速 ENA
//#define MotorRcountA 19 //编码器A 中断号：3
//#define MotorRcountB 18 //编码器B 中断号：2

//电机初始化
void Motor_Init(){
  //左电机
  pinMode(MotorLpin1,OUTPUT);
  pinMode(MotorLpin2,OUTPUT);
  pinMode(MotorLpwm,OUTPUT);
  //pinMode(MotorLcountA,INPUT);
  //pinMode(MotorLcountB,INPUT);
  
  //右电机
  pinMode(MotorRpin1,OUTPUT);
  pinMode(MotorRpin2,OUTPUT);
  pinMode(MotorRpwm,OUTPUT);
  //pinMode(MotorRcountA,INPUT);
  //pinMode(MotorRcountB,INPUT);
  
}

/**************************************************************************
  函数功能：小车向前移动
  入口参数：左右轮PWM值
  返回  值：无
**************************************************************************/
void Goforward(){
  //左电机
  digitalWrite(MotorLpin1,LOW);
  digitalWrite(MotorLpin2,HIGH);
  analogWrite(MotorLpwm,40);
  
  //右电机
  digitalWrite(MotorRpin1,HIGH);
  digitalWrite(MotorRpin2,LOW);
  analogWrite(MotorRpwm,200);
}

/**************************************************************************
  函数功能：小车向后移动
  入口参数：左右轮PWM值
  返回  值：无
**************************************************************************/
void Goback(){
  //左电机
  digitalWrite(MotorLpin1,HIGH);
  digitalWrite(MotorLpin2,LOW);
  analogWrite(MotorLpwm,150);
  
  //右电机
  digitalWrite(MotorRpin1,LOW);
  digitalWrite(MotorRpin2,HIGH);
  analogWrite(MotorRpwm,40);
}

void Stopcar(){
  //左电机
  digitalWrite(MotorLpin1,LOW);
  digitalWrite(MotorLpin2,LOW);
  analogWrite(MotorLpwm,0);
  
  //右电机
  digitalWrite(MotorRpin1,LOW);
  digitalWrite(MotorRpin2,LOW);
  analogWrite(MotorRpwm,0);
}

//原地左转
void TurnL_In_Situ(){
  
}

//原地右转
void TurnR_In_Situ(){
  
}

void setup() {
  // put your setup code here, to run once:
  Motor_Init();
  Goforward();
  //Serial.begin(9600);
}

void loop() {
  // put your main code here, to run repeatedly:
  /*
  Serial.println(TCCR0A, HEX);
  Serial.println(TCCR1A, HEX);
  Serial.println(TCCR2A, HEX);
  Serial.println(TCCR3A, HEX);
  Serial.println(TCCR4A, HEX);
  Serial.println(TCCR5A, HEX);
  delay(3000);
  */
}`

per1234, 2020-06-10:
Closing as duplicate of https://github.com/arduino/ArduinoCore-avr/issues/339

Please see that issue for explanation, workarounds, and updates about the resolution of this bug.

closed 2020-06-10 by per1234

cbirdfly, 2020-06-12:
> Closing as duplicate of arduino/ArduinoCore-avr#339

Thank you.