arduino/Arduino #10306: Colour of warning messages

HAJM, 2020-06-03:
Would it be possible to make the colour of warning messages different from the colour of error messages? Say yellow instead of red. I often get warnings about things in third party libraries that don't prevent them from working ok, but because they look the same as errors as they flash past me on the screen, I have to scroll back and look at them.  I know this is not a big problem, but if it was easy to fix in the next release, that would be nice.

feikname, 2020-06-09:
Also mentioned in #7451 (different issue though, so this should remain open)

matthijskooijman said in https://github.com/arduino/Arduino/issues/7451#issuecomment-387749315 that "The problem there is that it is hard to reliably decide whether something is an error, warning, etc." since everything is printed to stderr. 

I think then perhaps an alternative could be made: (i'm no expert in GGC internals this could be a bad suggestion)
 * Add a more heavyweight compile option for debugging purposes which compiles the code _two_ times (the second time preferably with some kind of cache if possible)
 1. Compile with warnings disabled
 2. Compile agin, with warnings enabled
 3. Do the stderr diff and you know which ones are warnings (or non-errors at least?) reliably.

This would be only for GGC cores of course, and more generic and easier to maintain than regexes at least.

The heavyweight compile option cpuld be triggered by clicking Shift+Compile or something of the sort. Maybe a preferences option also.

per1234, 2020-10-05:
>since the problems is third-party library warnings maybe add preference to disable warning output for library files?

Related: https://github.com/arduino/arduino-builder/issues/291