arduino/Arduino #10310: Pinout diagram Nano Every, pin D2 not correct.

Wild-duck-11, 2020-06-04:
On the pinout diagram of the Nano Every (https://content.arduino.cc/assets/Pinout-NANOevery_latest.pdf),
there is a "~" sign on the D2 pin. This sign is not correct. D2 is just a digital pin, not a PWM pin. On the Tech specs page and also in the schematic of the board, the PWM pins and pin D2 are indicated correctly.

prajwalraju, 2020-06-05:
I have removed the "~" symbol on the D2 pin
there are 2 files one is uncompressed and the other one file size is the same as the source

[Pinout-NANOevery_latest_uncompressed.pdf](https://github.com/arduino/Arduino/files/4736318/Pinout-NANOevery_latest_uncompressed.pdf)
[Pinout-NANOevery_latest.pdf](https://github.com/arduino/Arduino/files/4736327/Pinout-NANOevery_latest.pdf)

FunzioneFabio, 2020-07-03:
Hello! Thank you so much @Wild-duck-11 for reporting this error, I have corrected and uploaded the pinout on the website:
https://content.arduino.cc/assets/Pinout-NANOevery_latest.pdf

closed 2020-07-03 by kengdahl