arduino/Arduino #10321: Flat out cant scroll through code in arduino ide

jmb-555, 2020-06-07:
Please read the Issues section of the Contributing Rules at the "Contributing" link to the right before submitting an issue report.
I have done many Arduino projects and this has never been an issue. Two finger scroll fails to work. After rebooting the issue had gone away only to immediately reappear. One time after rebooting, the ability to two finger scroll was working but was unusably sensitive. It scrolled to the bottom of the page immediately. It currently just wont let me scroll at all.

Avamander, 2020-06-08:
This is most likely something wrong with your touchpad's drivers or configuration. Arduino IDE isn't the cause or only thing affected here.

closed 2020-12-24 by per1234

jmb-555, 2020-12-24:
Forget about it. Is fine. Just checked and its working. Like the other 10
programs I use, any problems that i have seem to just magically disappear
as soon as i write something about it on the website.