arduino/Arduino #10326: Autoformat should remove spaces after (if, while, switch) statement conditional

Avamander, 2020-06-08:
For example when the code looks like this:
```
          if (a != b)          {
            return;
          }
```

It should be turned into:
```
          if (a != b) {
            return;
          }
```

Same applies to switches:
```
switch (character)          {
```

Should be turned into:
```
switch (character) {
```

fabiuz7, 2021-03-25:
Do you know if Astyle is abandoned? I have looked at the repo but I don't see any activity in the past 2 years...

per1234, 2021-05-25:
Hi @Avamander thanks for your interest in improving the code formatting capabilities of the IDE!

There is some good news related to this feature request. https://github.com/arduino/Arduino/pull/11543 will change the Arduino IDE from using the Artistic Style formatter to [ClangFormat](https://clang.llvm.org/docs/ClangFormat.html), which provides just the sort of behavior you requested here.

ClangFormat is very popular and actively developed, and even suggested by the Artistic Style developer at the post @fabiuz7 kindly shared.[this statement](https://sourceforge.net/p/astyle/bugs/548/) where they state that they are no longer able to play as active a role in developing Artistic Style.

You can give the test build for the ClangFormat/Arduino IDE integration a try. The downloads are available here:
https://github.com/arduino/Arduino/pull/11543#issuecomment-845917365
If you do try it out, please let us know how it goes!

closed 2022-11-28 by Avamander