arduino/Arduino #10328: [Library manager] Add Library

ghlee-devel, 2020-06-09:
Please add this library to the Library manager
https://github.com/ghlee-devel/SoftPWM
Thanks,

per1234, 2020-06-10:
Hi @ghlee-devel. Arduino has a policy that 3rd party libraries with a library.properties `name` value starting with the word "Arduino" (or any capitalization thereof) will no longer be accepted for inclusion in the Arduino Library manager index. The reason is that Arduino wants to reserve these names for their own official libraries.

If you want your library to be added to the Library Manager index, you will need to change the `name` value in library properties to one that is compliant with the Arduino Library Specification:
https://arduino.github.io/arduino-cli/library-specification/#libraryproperties-file-format
Be aware that the name "softpwm" is already taken.

Then update the `version` value in library.properties and create a new release/tag, or else redo the existing release/tag.

Once you have done this, please comment here so we'll know we can proceed.

More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

per1234, 2020-08-04:
Closing due to lack of response.

@ghlee-devel if you every get around to making your library compliant with the requirements for addition to the Arduino Library Manager index, feel free to comment here and I'll be happy to re-open this issue.

closed 2020-08-04 by per1234