arduino/Arduino #10329: use external editor by default: how to link to a program?

dsyleixa, 2020-06-09:
in preferences there is an option "use external editor": 
but there a program link is missing, which editor to use instead by default (e.g., Notepad++)
(because the find + search/replace features in the Arduino editor are still mess...!)

matthijskooijman, 2020-06-09:
The "Use external editor" does not start the editor for you, you must do that manually.

The option only ensures that the internal editor is disabled, to prevent editing the same file twice (and to prevent that unsaved changes are present in the Arduino IDE, which are always used instead of the files on disk otherwise). It also ensures that files are reloaded when they are externally modified (this was historically needed to ensure the right version was compiled, but now with arduino-builder, the version on disk is always used if the IDE has no unsaved changes).

So I think there's no bug here, but maybe the wording or documentation around this feature could be improved?

dsyleixa, 2020-06-09:
yes, both an improvement of documentation and an improvement by direct embedding of an external monitor then to start it automatically, as a 2nd step please!

per1234, 2020-06-09:
We already have a feature request to allow an external editor application to be associated with the IDE:
https://github.com/arduino/Arduino/issues/8768

Let's confine any further discussion about that feature to the original issue. We can make this issue about documenting the "Use external editor" feature. As far as I know, this feature is undocumented. The feature periodically causes confusion with the IDE users who enable the feature and then can't figure out why the IDE's editor pane is disabled for input (e.g., https://github.com/arduino/Arduino/issues/7351).

The appropriate place to add this documentation would be https://www.arduino.cc/en/Guide/Environment

dsyleixa, 2020-06-12:
actually I run into this issue because  I wanted to use an external editor automatically by enabling the option "use external editor":
but there a program link is missing, which editor to use instead by default.
Just disabling the internal editor would be no option for me at all.

So it's not just a matter of a missing documentation but also of a link to the preferred external editor.

per1234, 2020-06-12:
As I said in my previous reply, there is a previous dedicated issue for that feature request: https://github.com/arduino/Arduino/issues/8768

Please confine any additional discussion on this issue to the topic of documenting the external editor feature.

dsyleixa, 2020-06-12:
Just a documentation about disabling is not my poiint, without providing a link to an external editor.
So I close this aa a duplicate of https://github.com/arduino/Arduino/issues/8768.

closed 2020-06-12 by dsyleixa