arduino/Arduino #10331: Board Manager Java Error

Sausebraus, 2020-06-09:
After i open the board manager i get this error:

 Fehler beim Herunterladen von http://downloads.arduino.cc/packages/package_index.json
java.lang.RuntimeException: java.lang.Exception: Fehler beim Herunterladen von http://downloads.arduino.cc/packages/package_index.json
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$0(ContributionManagerUI.java:151)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: Fehler beim Herunterladen von http://downloads.arduino.cc/packages/package_index.json
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:131)
	at cc.arduino.contributions.packages.ContributionInstaller.download(ContributionInstaller.java:334)
	at cc.arduino.contributions.packages.ContributionInstaller.downloadIndexAndSignature(ContributionInstaller.java:309)
	at cc.arduino.contributions.packages.ContributionInstaller.updateIndex(ContributionInstaller.java:287)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$0(ContributionManagerUI.java:147)
	... 1 more
Caused by: java.net.UnknownHostException: downloads.arduino.cc
	at java.net.AbstractPlainSocketImpl.connect(AbstractPlainSocketImpl.java:184)
	at java.net.PlainSocketImpl.connect(PlainSocketImpl.java:172)
	at java.net.SocksSocketImpl.connect(SocksSocketImpl.java:392)
	at java.net.Socket.connect(Socket.java:589)
	at sun.net.NetworkClient.doConnect(NetworkClient.java:175)
	at sun.net.www.http.HttpClient.openServer(HttpClient.java:463)
	at sun.net.www.http.HttpClient.openServer(HttpClient.java:558)
	at sun.net.www.http.HttpClient.<init>(HttpClient.java:242)
	at sun.net.www.http.HttpClient.New(HttpClient.java:339)
	at sun.net.www.http.HttpClient.New(HttpClient.java:357)
	at sun.net.www.protocol.http.HttpURLConnection.getNewHttpClient(HttpURLConnection.java:1202)
	at sun.net.www.protocol.http.HttpURLConnection.plainConnect0(HttpURLConnection.java:1181)
	at sun.net.www.protocol.http.HttpURLConnection.plainConnect(HttpURLConnection.java:1032)
	at sun.net.www.protocol.http.HttpURLConnection.connect(HttpURLConnection.java:966)
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:170)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:128)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:129)
	... 5 more
java.lang.NullPointerException
java.lang.RuntimeException: java.lang.NullPointerException
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$0(ContributionManagerUI.java:151)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.NullPointerException
	at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:123)
	at processing.app.BaseNoGui.initPackages(BaseNoGui.java:487)
	at processing.app.Base$9.onIndexesUpdated(Base.java:1377)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$0(ContributionManagerUI.java:149)
	... 1 more
java.lang.NullPointerException
java.lang.RuntimeException: java.lang.NullPointerException
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$1(ContributionManagerUI.java:176)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.NullPointerException
	at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:123)
	at processing.app.BaseNoGui.initPackages(BaseNoGui.java:487)
	at processing.app.Base$9.onIndexesUpdated(Base.java:1377)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$1(ContributionManagerUI.java:174)
	... 1 more
java.lang.NullPointerException
java.lang.RuntimeException: java.lang.NullPointerException
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$1(ContributionManagerUI.java:176)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.NullPointerException
	at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:123)
	at processing.app.BaseNoGui.initPackages(BaseNoGui.java:487)
	at processing.app.Base$9.onIndexesUpdated(Base.java:1377)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$1(ContributionManagerUI.java:174)
	... 1 more
java.lang.NullPointerException
	at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:123)
	at processing.app.BaseNoGui.initPackages(BaseNoGui.java:487)
	at processing.app.Base.openBoardsManager(Base.java:1395)
	at processing.app.Base$10.actionPerformed(Base.java:1418)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.AbstractButton.doClick(AbstractButton.java:376)
	at javax.swing.plaf.basic.BasicMenuItemUI.doClick(BasicMenuItemUI.java:833)
	at javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(BasicMenuItemUI.java:877)
	at java.awt.Component.processMouseEvent(Component.java:6533)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6298)
	at java.awt.Container.processEvent(Container.java:2236)
	at java.awt.Component.dispatchEventImpl(Component.java:4889)
	at java.awt.Container.dispatchEventImpl(Container.java:2294)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4888)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4525)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4466)
	at java.awt.Container.dispatchEventImpl(Container.java:2280)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:90)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.awt.EventQueue$4.run(EventQueue.java:729)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

If I close Arduino and try to start it again, it wont open. After deleting the Arduino15 folder it starts working again. Reinstalled many times, two different Win10 PCs. 
Arduino UNO Wifi Rev2 board

feetball, 2020-06-09:
Same issue on Windows 10.  Have to delete C:\Users\<user>\AppData\Local\Arduino15 to get it to run again, however, like niclas0219 said, running the boards manager breaks it again

Running the nightly build seems to work.

k0mmsussert0d, 2020-06-09:
Can confirm. Version 1.8.12-1 on both OpenJDK 8 and 11.

877dev, 2020-06-09:
Same here, two machines. So glad it's not just me :)

AngryArduino, 2020-06-09:
Same. Two devices, even when I use the libraries manager I have the same issue.

arfr, 2020-06-09:
Same here. With 3 devices =D Win 10 (Desktop, Notebook) and Fedora 31 ... please help! need only to change the delay for my esp32 device )))

arfr, 2020-06-09:
> Same issue on Windows 10. Have to delete C:\Users\\AppData\Local\Arduino15 to get it to run again, however, like niclas0219 said, running the boards manager breaks it again
> 
> Running the nightly build seems to work.

Yep, beta and nightly works!

Nathan-ma, 2020-06-09:
same here! lost a day of work with this bug and it is affecting vscode extension.

KennethLavrsen, 2020-06-09:
I also have this problem. I run on Ubuntu Linux with a mature installation and I did not upgrade anything or install any new boards today.
It worked around 6 pm Central European Time. It was broken at 10 pm. Arduino will not start at all.

If I reinstall a fresh Arduino I can start it. The minute I open board manager without having added anything the java script errors occurs and you can no longer start Arduino.

It must be some garbage that someone checked in that Arduino gets from the internet when you start board manager that causes ther crash

KCole313, 2020-06-09:
From what I can tell it seems like there is an issue with the package_index.json file that is being fetched when the package manager is opened. If I delete just that file it starts up and works fine until I open the package manager again and the file is downloaded again.

svenjie, 2020-06-09:
I'm getting the same bug.

svenjie, 2020-06-09:
java.lang.NullPointerException
java.lang.RuntimeException: java.lang.NullPointerException
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$1(ContributionManagerUI.java:150)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.NullPointerException
	at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:134)
	at processing.app.BaseNoGui.initPackages(BaseNoGui.java:483)
	at processing.app.Base$9.onIndexesUpdated(Base.java:1400)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$1(ContributionManagerUI.java:148)

aidzer0, 2020-06-09:
1.9 beta is working for me, I am getting errors when opening Baard manager, but it still works;
```
Index error: could not find referenced tool name=openocd version=0.10.0-arduino12 packager=arduino
Index error: could not find referenced tool name=dfu-util version=0.9.0-arduino2 packager=arduino
```

facchinm, 2020-06-10:
We had a problem on our package_index.json that caused the error and prevented the IDE to start.
Now the package_index has been fixed, but to recover a working installation you must delete once again the package_index.json with the cache folder and restart the IDE as usual.

Windows:
* remove file C:\Users\YourUsername\AppData\Local\Arduino15\package_index.json
* remove folder C:\Users\YourUsername\AppData\Local\Arduino15\cache

Mac:
* remove file /Users/YourUsername/Library/Arduino15/package_index.json
* remove folder /Users/YourUsername/Library/Arduino15/cache

Linux:
* remove file /home/YourUsername/.arduino15/package_index.json
* remove folder /home/YourUsername/.arduino15/cache

Please note that the problem that prevents the IDE to start has already been fixed and it's ready for the next release, that's the reason why the Nightly/Beta Builds are not affected, and also the reason why we did not detect this problem earlier, sorry about that!

closed 2020-06-10 by facchinm

kfezer, 2020-09-14:
This still seems to be happening with the library manager:

` no protocol: 
java.lang.RuntimeException: java.net.MalformedURLException: no protocol: 
	at cc.arduino.contributions.libraries.ui.LibraryManagerUI.lambda$onInstallPressed$4(LibraryManagerUI.java:248)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.net.MalformedURLException: no protocol: 
	at java.net.URL.<init>(URL.java:611)
	at java.net.URL.<init>(URL.java:508)
	at java.net.URL.<init>(URL.java:457)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:64)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:60)
	at cc.arduino.contributions.libraries.LibraryInstaller.performInstall(LibraryInstaller.java:155)
	at cc.arduino.contributions.libraries.LibraryInstaller.install(LibraryInstaller.java:125)
	at cc.arduino.contributions.libraries.ui.LibraryManagerUI.lambda$onInstallPressed$4(LibraryManagerUI.java:237)
	... 1 more
`

I tried the above steps of deleting the cache and including a fresh install and still cannot get around this error.

oldjiberjaber, 2020-10-12:
Same here, tried the suggested deletion but  I am now stuck being unable to progress some projects...
(Arduino 1.8.13)

Index error: could not find referenced tool name=avr-gcc version=4.8.1-arduino5 packager=arduino
 no protocol: 
java.lang.RuntimeException: java.net.MalformedURLException: no protocol: 
	at cc.arduino.contributions.libraries.ui.LibraryManagerUI.lambda$onInstallPressed$4(LibraryManagerUI.java:248)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.net.MalformedURLException: no protocol: 
	at java.net.URL.<init>(URL.java:593)
	at java.net.URL.<init>(URL.java:490)
	at java.net.URL.<init>(URL.java:439)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:64)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:60)
	at cc.arduino.contributions.libraries.LibraryInstaller.performInstall(LibraryInstaller.java:155)
	at cc.arduino.contributions.libraries.LibraryInstaller.install(LibraryInstaller.java:125)
	at cc.arduino.contributions.libraries.ui.LibraryManagerUI.lambda$onInstallPressed$4(LibraryManagerUI.java:237)
	... 1 more