arduino/Arduino #10332: Editor Failing to start

rjs5327, 2020-06-09:
So i am running a Win 10 machine and on 2 of them i get the same problem, including both the desktop and portable version.  This issue just arose all was working and then i got the normal double port error where it shows say COM 8 twice and wont do anything so i rebooted the Editor and it wont start.  

Here is the debug print out:
Loading configuration...
Initializing packages...
java.lang.NullPointerException
        at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:134)
        at processing.app.BaseNoGui.initPackages(BaseNoGui.java:483)
        at processing.app.Base.<init>(Base.java:273)
        at processing.app.Base.main(Base.java:150)

This all happened after trying to connect to some MKR's having trouble i think they are too close and the WIFI is screwing up.  Since the boards randomly reboot and LED_BUILTIN pulses slow.  So i been messing with positioning the boards and pulling things out of the code to find the issue.  And my machines have locked up with the above.  So i tried portable doesnt work.  Tried uninstalling and reinstalling does not work.  So im stuck

rjs5327, 2020-06-09:
Interesting I tried the 1.9 Beta and it will run in portable but the 1.8.12 will not.  I saw in the release notes that there was a change in the JAVA maybe that is where the issue lays

niclas0219, 2020-06-09:
Hi!
I get the same thing. When I delete the package_index.json file it starts again.
But when using the board manager i get a null pointer exception and then it stops working again. I'm thinking something is broken in the standard board index that the IDE downloads from somewhere.

WillEccles, 2020-06-09:
I have encountered the same issue. This seems to be a bit of sticking point for Arduino, as I have found multiple other old issues about very similar issues.

niclas0219, 2020-06-09:
I can also confirm that the 1.9 beta is working!

WillEccles, 2020-06-09:
Seconded, 1.9 works here.

CptAdventure, 2020-06-09:
Yup, I just upgraded to 1.8.12 after running 1.8.5 flawlessly. 
When I try to start the GUI, I get the splash screen with "Initializing Packages..." and then the splash screen goes away and no editor. Nana. Dead in the water.
I going to try 1.9 and see if any changes.

877dev, 2020-06-09:
Exactly the same here, occured on both my separate machines. Driving me mad.
Happens when going to board manager:

`java.lang.NullPointerException
java.lang.RuntimeException: java.lang.NullPointerException
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$1(ContributionManagerUI.java:150)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.NullPointerException
	at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:134)
	at processing.app.BaseNoGui.initPackages(BaseNoGui.java:483)
	at processing.app.Base$9.onIndexesUpdated(Base.java:1400)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$1(ContributionManagerUI.java:148)
	... 1 more
java.lang.NullPointerException
	at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:134)
	at processing.app.BaseNoGui.initPackages(BaseNoGui.java:483)
	at processing.app.Base.openBoardsManager(Base.java:1418)
	at processing.app.Base$10.actionPerformed(Base.java:1441)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.AbstractButton.doClick(AbstractButton.java:376)
	at javax.swing.plaf.basic.BasicMenuItemUI.doClick(BasicMenuItemUI.java:842)
	at javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(BasicMenuItemUI.java:886)
	at java.awt.Component.processMouseEvent(Component.java:6539)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6304)
	at java.awt.Container.processEvent(Container.java:2239)
	at java.awt.Component.dispatchEventImpl(Component.java:4889)
	at java.awt.Container.dispatchEventImpl(Container.java:2297)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4904)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4535)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4476)
	at java.awt.Container.dispatchEventImpl(Container.java:2283)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:760)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:84)
	at java.awt.EventQueue$4.run(EventQueue.java:733)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:730)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)`

AngryArduino, 2020-06-09:
Same,
java.lang.NullPointerException
java.lang.RuntimeException: java.lang.NullPointerException
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$1(ContributionManagerUI.java:150)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.NullPointerException
	at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:124)
	at processing.app.BaseNoGui.initPackages(BaseNoGui.java:484)
	at processing.app.Base$9.onIndexesUpdated(Base.java:1376)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$1(ContributionManagerUI.java:148)
	... 1 more
java.lang.NullPointerException
	at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:124)
	at processing.app.BaseNoGui.initPackages(BaseNoGui.java:484)
	at processing.app.Base.openBoardsManager(Base.java:1394)
	at processing.app.Base$10.actionPerformed(Base.java:1417)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.AbstractButton.doClick(AbstractButton.java:376)
	at javax.swing.plaf.basic.BasicMenuItemUI.doClick(BasicMenuItemUI.java:842)
	at javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(BasicMenuItemUI.java:886)
	at java.awt.Component.processMouseEvent(Component.java:6539)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6304)
	at java.awt.Container.processEvent(Container.java:2239)
	at java.awt.Component.dispatchEventImpl(Component.java:4889)
	at java.awt.Container.dispatchEventImpl(Container.java:2297)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4904)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4535)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4476)
	at java.awt.Container.dispatchEventImpl(Container.java:2283)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:760)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:84)
	at java.awt.EventQueue$4.run(EventQueue.java:733)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:730)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

pfeigl, 2020-06-09:
Can also reproduce this. 

* Fresh install of 1.8.12
* Add `https://arduino.esp8266.com/stable/package_esp8266com_index.json` in board manager
* Go to board manager and get `NullPointerException`

After that, the editor cannot start any longer and fails with
```
java.lang.NullPointerException
        at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:134)
        at processing.app.BaseNoGui.initPackages(BaseNoGui.java:483)
        at processing.app.Base.<init>(Base.java:273)
        at processing.app.Base.main(Base.java:150)
```

Perhaps something specific with the `esp8266` json?

Erwin-66, 2020-06-09:
Same problem here. Tried 1.8.12 and earlier versions. Also removed \AppData\Local\Arduino15 folder and installed again. Then I can start the editor but after first board manager entrance I get the java.lang.nullpointexception error. After closing the editor it does not start up anymore.

Danixu, 2020-06-09:
> Can also reproduce this.
>
> * Fresh install of 1.8.12
> * Add `https://arduino.esp8266.com/stable/package_esp8266com_index.json` in board manager
> * Go to board manager and get `NullPointerException` [...]

Nope, I've cleaned everything and installed an older version that was working (1.8.10), and same error. Maybe is something related with packages index, because is working fine until I enter into cards manager. The I got the already mentiones errors and never again starts until I remove the package_index.json files.

Best regards.

APJoshi-Projects, 2020-06-09:
Same problem here with arduino ide 1.8.12: uploaded FSBrowser example code to nodemcu-esp8266 and closed the ide. When i try to start the ide now it shows Initializing Packages... and crashes.

zeucxb, 2020-06-09:
@APJoshi-Projects same here.

luk3skyw4lker, 2020-06-09:
same here, trying 1.9 now

Petorrr, 2020-06-09:
Same problem here, worked flawlessly with multiple boards and projects ever since I installed 1.8.12. Tonight it stopped working when I opened the Boards Manager. Re-install doesn't work, and neither the Nightly build. Same problem also now on my other computer! You can delete the Arduino15 folder in Appdata, and it will start once, but when the indexes are generated the next time it won't start anymore. When it starts and you open the Boards Manager it acts really weird and a nullpointer exception comes in the bottom of the window.

luk3skyw4lker, 2020-06-09:
1.9 opens here, board manager works fine, but returns:

`
Index error: could not find referenced tool name=openocd version=0.10.0-arduino12 packager=arduino
Index error: could not find referenced tool name=dfu-util version=0.9.0-arduino2 packager=arduino
Index error: could not find referenced tool name=openocd version=0.10.0-arduino12 packager=arduino
Index error: could not find referenced tool name=dfu-util version=0.9.0-arduino2 packager=arduino
Index error: could not find referenced tool name=openocd version=0.10.0-arduino12 packager=arduino
Index error: could not find referenced tool name=dfu-util version=0.9.0-arduino2 packager=arduino
Index error: could not find referenced tool name=openocd version=0.10.0-arduino12 packager=arduino
Index error: could not find referenced tool name=dfu-util version=0.9.0-arduino2 packager=arduino
Index error: could not find referenced tool name=openocd version=0.10.0-arduino12 packager=arduino
Index error: could not find referenced tool name=dfu-util version=0.9.0-arduino2 packager=arduino
`

obdevel, 2020-06-09:
There have been multiple posts on Reddit about this in the past few hours. Has some random Windows update triggered this behaviour ? fwiw my Mac is fine.

osvaldo13576, 2020-06-09:
I have the same issue in Archlinux

l1ttled3vil, 2020-06-09:
> 
> 
> There have been multiple posts on Reddit about this in the past few hours. Has some random Windows update triggered this behaviour ? fwiw my Mac is fine.

My Arduino IDE 1.8.12 has been running fine for a long time on Windows 10, and this morning it just won't boot. I've not changed anything on my PC so it must be an update that has automatically gone through somewhere...

l1ttled3vil, 2020-06-09:
A workaround for now is to delete the Arduino folder under (C:\Users\user\AppData\Local). The IDE will boot, but you won't have your additional boards available.

avillacis, 2020-06-09:
> A workaround for now is to delete the Arduino folder under (C:\Users\user\AppData\Local). The IDE will boot, but you won't have your additional boards available.

Just delete the package_index.json and package_index.json.sig files inside said directory. Will preserve current board support, but you will not be able to install or update boards until this is fixed.

jPerotto, 2020-06-10:
https://forum.arduino.cc/index.php?topic=689307.msg4635898#msg4635898

temporary solution:

1) Disable machine's networking.
2) Copy the .../arduino15 directory to,say, arduino15.backup
3) Remove the files package_index.json and package_index.json.sig from the ..../arduino15
4) Start the IDE. It should start without network access. Go to Preferences, disable "check for updates at start". Stop the IDE.
5) Start networking
6) Start IDE. It starts.
7) Avoid any touching of the network resources! The board_manager or library_manager buggers the package_index again and again.

But add firewall exit rules from arduino.exe, arduino_debug.exe, arduino-builder.exe and arduino-cli.exe.

jonasbradstreet, 2020-06-10:
Picked up JAVA_TOOL_OPTIONS:
Set log4j store directory /home/programming/.arduino15
Gtk-Message: 18:58:14.166: Failed to load module "canberra-gtk-module"
java.lang.NullPointerException
        at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:134)
        at processing.app.BaseNoGui.initPackages(BaseNoGui.java:483)
        at processing.app.Base.<init>(Base.java:273)
        at processing.app.Base.main(Base.java:150)

mjs513, 2020-06-10:
Having exactly the same issue:
https://forum.arduino.cc/index.php?topic=671157.new#new since this morning

ubdussamad, 2020-06-10:
> Can also reproduce this.
>
>     * Fresh install of 1.8.12
>
>     * Add `https://arduino.esp8266.com/stable/package_esp8266com_index.json` in board manager
>
>     * Go to board manager and get [...]

HI, it's not just limited to the ESP family.

chefer, 2020-06-10:
https://stackoverflow.com/questions/62290881/arduino-ide-win10-crashes-when-i-launch-the-program

chrissbarr, 2020-06-10:
I'm seeing this as well. Fresh install of 1.8.12, opening the Boards Manager downloads the default packages then throws a `NullPointerException`. This is without any extra packages added, just the stock Arduino ones.

nateimig18, 2020-06-10:
> https://forum.arduino.cc/index.php?topic=689307.msg4635898#msg4635898
>
> temporary solution:
>
> 1. Disable machine's networking.
> 2. Copy the .../arduino15 directory to,say, arduino15.backup
> 3. Remove [...]

Thanks, I had narrowed it down to these two files but they kept cropping back up. Why do the firewall exit rules need to be in place?

JCT250, 2020-06-10:
> https://forum.arduino.cc/index.php?topic=689307.msg4635898#msg4635898
>
> temporary solution:
>
> 1. Disable machine's networking.
> 2. Copy the .../arduino15 directory to,say, arduino15.backup
> 3. Remove [...]

If you do this you still won't have access to any boards aside from teh default ones though yea? (at least that is my experience)

MatthewReed303, 2020-06-10:
I just let windows do an update and exact same issues as everyone else. I'm going to roll back the windows update. I have been using Arduino fine most of the day and windows asks to restart to install the update, let it do it's thing and reboot, go to open Arduino again and can't get past splash screen. I tried all of the suggested stuff but nothing works. I need additional boards manager as I'm using the stm32

KernelErr, 2020-06-10:
I met the same error when I have a Windows update. Deleted `package_index.json` and solved this problem. But still don't know how it happens.

```
正在加载配置...
正在初始化包...
java.lang.NullPointerException
        at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:134)
        at processing.app.BaseNoGui.initPackages(BaseNoGui.java:483)
        at processing.app.Base.<init>(Base.java:273)
        at processing.app.Base.main(Base.java:150)
```

javierzebadua, 2020-06-10:
I have the same error with boards manager, hours ago it use to be fine, but now arduino doesnt start, when i delete `package_index.json` and `package_index.json.sig`, it starts, but when i open boards manager, it crash with these errors
`java.lang.NullPointerException
java.lang.RuntimeException: java.lang.NullPointerException
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$1(ContributionManagerUI.java:150)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.NullPointerException
	at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:134)
	at processing.app.BaseNoGui.initPackages(BaseNoGui.java:483)
	at processing.app.Base$9.onIndexesUpdated(Base.java:1400)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$1(ContributionManagerUI.java:148)
	... 1 more
Exception in thread "AWT-EventQueue-0" java.lang.NullPointerException
	at cc.arduino.contributions.packages.ui.ContributedPlatformTableCellJPanel.update(ContributedPlatformTableCellJPanel.java:222)
	at cc.arduino.contributions.packages.ui.ContributedPlatformTableCellRenderer.getTableCellRendererComponent(ContributedPlatformTableCellRenderer.java:47)
	at javax.swing.JTable.prepareRenderer(JTable.java:5729)
	at javax.swing.plaf.basic.BasicTableUI.paintCell(BasicTableUI.java:2114)
	at javax.swing.plaf.basic.BasicTableUI.paintCells(BasicTableUI.java:2016)
	at javax.swing.plaf.basic.BasicTableUI.paint(BasicTableUI.java:1812)
	at javax.swing.plaf.ComponentUI.update(ComponentUI.java:161)
	at javax.swing.JComponent.paintComponent(JComponent.java:780)
	at javax.swing.JComponent.paint(JComponent.java:1056)
	at javax.swing.JComponent.paintChildren(JComponent.java:889)
	at javax.swing.JComponent.paint(JComponent.java:1065)
	at javax.swing.JViewport.paint(JViewport.java:728)
	at javax.swing.JComponent.paintChildren(JComponent.java:889)
	at javax.swing.JComponent.paint(JComponent.java:1065)
	at javax.swing.JComponent.paintToOffscreen(JComponent.java:5210)
	at javax.swing.RepaintManager$PaintManager.paintDoubleBuffered(RepaintManager.java:1579)
	at javax.swing.RepaintManager$PaintManager.paint(RepaintManager.java:1502)
	at javax.swing.RepaintManager.paint(RepaintManager.java:1272)
	at javax.swing.JComponent._paintImmediately(JComponent.java:5158)
	at javax.swing.JComponent.paintImmediately(JComponent.java:4969)
	at javax.swing.RepaintManager$4.run(RepaintManager.java:831)
	at javax.swing.RepaintManager$4.run(RepaintManager.java:814)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at javax.swing.RepaintManager.paintDirtyRegions(RepaintManager.java:814)
	at javax.swing.RepaintManager.paintDirtyRegions(RepaintManager.java:789)
	at javax.swing.RepaintManager.prePaintDirtyRegions(RepaintManager.java:738)
	at javax.swing.RepaintManager.access$1200(RepaintManager.java:64)
	at javax.swing.RepaintManager$ProcessingRunnable.run(RepaintManager.java:1732)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:109)
	at java.awt.WaitDispatchSupport$2.run(WaitDispatchSupport.java:190)
	at java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:235)
	at java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:233)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.awt.WaitDispatchSupport.enter(WaitDispatchSupport.java:233)
	at java.awt.Dialog.show(Dialog.java:1084)
	at java.awt.Component.show(Component.java:1671)
	at java.awt.Component.setVisible(Component.java:1623)
	at java.awt.Window.setVisible(Window.java:1014)
	at java.awt.Dialog.setVisible(Dialog.java:1005)
	at processing.app.Base.openBoardsManager(Base.java:1414)
	at processing.app.Base$10.actionPerformed(Base.java:1441)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.AbstractButton.doClick(AbstractButton.java:376)
	at javax.swing.plaf.basic.BasicMenuItemUI.doClick(BasicMenuItemUI.java:842)
[... 141 lines cut ...]
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4904)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4535)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4476)
	at java.awt.Container.dispatchEventImpl(Container.java:2283)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4711)
[... cut ...]
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:730)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)`

snowe2010, 2020-06-10:
Also seeing this issue on Mac, but unsure how to debug it, as there is no error when starting from the command line, it just closes after the splash screen.

javierzebadua, 2020-06-10:
Yes, its because when Arduino starts, it tries to update the boards index, but it can't do it, so it close without any window, I don't how Macs works but, you need to find and delete the package_index.json to open arduino

engynnadmin, 2020-06-10:
I am having the same issue, came out of nowhere. Glad to have found this thread after spending hours. I will go to 1.9

OrgiaMode, 2020-06-10:
I guess I'll share my frustration too. 1.8.12 stopped opening this evening and after a few hours of googling, I came across the above error after running debug in the cmd. All I can add to the discussion is that I am sick of these things happening at random times for no good reason.

Rocova98, 2020-06-10:
I have the same issue, I can work work with any board of Arduino but when I try to download another package it doesn´t work so I try 1.9 and also I can´t download any board through Preferences

dominicdj37, 2020-06-10:
same issue in Ubuntu. Was working fine till yesterday until after installing an esp board lib from boards manager.

MatthewReed303, 2020-06-10:
So 1.9 beta worked until closed down and then try to reopen and same as 1.8.12. I tried deleting the package index files and no luck. Is this all related to windows update or just randomly?

Guntanuth, 2020-06-10:
I have the same on Window10 non-admin , I tried 1.8.8 ,1.8.9 .1.8.12 NOT work , but 1.6.9 is WORK.
window 've just updated this morning.
C:\Users\guntanuh\Documents\Arduino_1.12>arduino_debug
Set log4j store directory C:\Users\guntanuh\AppData\Local\Arduino15
Loading configuration...
Initializing packages...
java.lang.NullPointerException
        at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:134)
        at processing.app.BaseNoGui.initPackages(BaseNoGui.java:483)
        at processing.app.Base.<init>(Base.java:273)
        at processing.app.Base.main(Base.java:150)

C:\Users\guntanuh\Documents\Arduino_1.12>

Danixu, 2020-06-10:
> So 1.9 beta worked until closed down and then try to reopen and same as 1.8.12. I tried deleting the package index files and no luck. Is this all related to windows update or just randomly?

Is not a Windows update related problem. There are people with Mac/Linux getting the same error, and I'm getting errors with packages index even in Eclipse (at least exlipse ignores those errors and still working), so there's no real problem with the IDE. I think that the problem is on the package index.

radiobla, 2020-06-10:
I have the same issue on Arch Linux. It was running flawlessly until a few hours ago.

**The following happens with Oracle JDK 8:**

`2020-06-10T06:43:39.231Z INFO c.a.c.p.ContributionInstaller:314 [ContributionManager Update Thread] Downloaded package index URL=[https://downloads.arduino.cc/packages/package_index.json]
2020-06-10T06:43:39.231Z INFO c.a.c.p.ContributionInstaller:324 [ContributionManager Update Thread] Check unknown files. Additional package index folder files=[package_index.json], Additional package index url downloaded=[]
java.lang.NullPointerException
java.lang.RuntimeException: java.lang.NullPointerException
        at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$1(ContributionManagerUI.java:150)
        at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.NullPointerException
        at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:134)
        at processing.app.BaseNoGui.initPackages(BaseNoGui.java:483)
        at processing.app.Base$9.onIndexesUpdated(Base.java:1400)
        at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$1(ContributionManagerUI.java:148)
        ... 1 more
`

`$ arduino
Picked up JAVA_TOOL_OPTIONS: 
Set log4j store directory /home/adon/.arduino15
java.lang.NullPointerException
        at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:134)
        at processing.app.BaseNoGui.initPackages(BaseNoGui.java:483)
        at processing.app.Base.<init>(Base.java:273)
        at processing.app.Base.main(Base.java:150)
`

This happens with Oracle JDK 11.0.7 (that was running until a few hours ago).

`WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by processing.app.linux.GTKLookAndFeelFixer (file:/usr/share/arduino/lib/arduino-core.jar) to field com.sun.java.swing.plaf.gtk.GTKLookAndFeel.styleFactory
WARNING: Please consider reporting this to the maintainers of processing.app.linux.GTKLookAndFeelFixer
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
java.lang.NullPointerException
        at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:134)
        at processing.app.BaseNoGui.initPackages(BaseNoGui.java:483)
        at processing.app.Base.<init>(Base.java:273)
        at processing.app.Base.main(Base.java:150)
`
In the later, it seems something related to GTK and with JDK 8 it complains about parsing package index JSON. 

Arduino 1.8.12 is unusable. It needs an upgrade soon.

I can't continue working on my prototypes until it's solved. 

What is incredible is that Arduino's web site doesn't mention a single word about the Java requirements, versions, etc. That's ridiculous.

I was about to buy another board and modules, but with these issues, I can't trust Arduino IDE anymore. At anytime it's going to stop working.

GregS1000RR, 2020-06-10:
Same here, ... spent entire evening de-installing - deleting  reinstalling 
... until came across this thread.

Thanks for having opened it ... fell less alone

radiobla, 2020-06-10:
A "workaround" would be deleting the ~/.arduino15 directory. The IDE will open again, but won't work as expected.

dergizzle, 2020-06-10:
Good to know I am not insane - cost me hours yesterday. The package_index.json is obviously the culprit, loading it will now result in the nullpointer exception and the subsequent refusal to startup the IDE. The workardounds mentioned sort of work, so does restoring a "good" copy of package_index.json from a backup hence it's something on their end.

Strange to see a nightly fix this, I would expect the fix to happen at the package_index.json side to ensure things still work for everybody .......

radiobla, 2020-06-10:
@dergizzle Yeah, what operating system are you running? which JDK/JRE?

You're not insane. Maybe it is who commited this code.

dergizzle, 2020-06-10:
I'm on Win 10 on all machines (1909 through the board) and I'm on the JRE that's bundled with the IDE. It's definately not an OS or Java related thing, too many people affected at the same time. Stuff like that is something central, so I was thinking certificates - but I'm pretty sure it's connected to the package_index.json. Or more specifically the the redesign of the board selector sub-menu...

(See [Picture](https://i.csm.sx/image.php?id=3c8a37524b06d14771d47734d5d02816_5ee0856f07a8e.png) )

radiobla, 2020-06-10:
Good point. I don't think they will look into this issue soon. So I have to stop everything I was doing.

It might be something with the JSON Parser too (it worth mention that I validated the JSON file and it's ok). There's a ".sig" (signature file), but I'm not sure it's related.

How I hate JVM!

mrguen, 2020-06-10:
Same here. Anyways Arduino with Arduino core STM32 was so slow since months I could hardly use it (compilation taking 20 minutes to compile the entire hundreds of Mo of libraries). Thanks I can't even install the boards definition now OMG.

lucasroth, 2020-06-10:
@dergizzle  hey, another not insane, just sleepless person here! win 10, arduino ide 1.8.10 everything worked fine till reboot of the PC and then the IDE stopped launching. Hours getting crazy about this issue, now can compile and upload deleting the package_index.json from folder: C:\Users\%my username%\AppData\Local\Arduino15  hope the fix is available soon, several hours wasted today! good night

GregS1000RR, 2020-06-10:
The only workaround which worked for me was to revert back to 1.8.10

Beta 1.9 isn't able to compile Tasmota project

yellowbluesky, 2020-06-10:
The hourly builds (I installed 4 June 2020 0:13:17 GMT) available [here](https://www.arduino.cc/en/Main/Software#hourly) are working for me.

I can close and reopen the IDE as many times as I want without.

Board manager and Library manager appear to work fine. I have only tested with the Arduino Mega 2560 core

darshit115, 2020-06-10:
I have uninstall 1.8.7 version and reinstall the 1.8.12 version but still i am getting error in Board manager "java.lang.NullPointerException".

S10143806H, 2020-06-10:
> Can also reproduce this.
>
> * Fresh install of 1.8.12
> * Add `https://arduino.esp8266.com/stable/package_esp8266com_index.json` in board manager
> * Go to board manager and get `NullPointerException` [...]

I encountered the same issue after trying to add Add another `.json` in the board manager

dylanpil, 2020-06-10:
I feel a little better, seeing that im not the only one having this issue.
I have been stuck on this for approx 4 hours, and it has really ruined my day, as I have projects to finish, and can't get anywhere.
I was about to format my computer and start again, thank goodness I see that its not just me.

I found all the same issues, but i am able to get the hourly IDE to open.
But this one has Index Error

I hope someone works out a proper solution soon.

Fernandoantvf, 2020-06-10:
> Just delete the package_index.json and package_index.json.sig files inside said directory. Will preserve current board support, but you will not be able to install or update boards until this is [...]

This solution worked for me!

VladoPortos, 2020-06-10:
Yesterday it was working ok, turned off pc and went sleep.. come back to turned on PC and not working IDE :-/ same  issue as above...

dylanpil, 2020-06-10:
@Fernandoantvf 
did you go to board manager?
as this is one part that just doesnt want to work

Fernandoantvf, 2020-06-10:
> @Fernandoantvf
> did you go to board manager?
> as this is one part that just doesnt want to work

Yes, after deleting the files  package_index.json and package_index.json.sig Arduino opened normally and all boards are available.

dylanpil, 2020-06-10:
@Fernandoantvf 
Interesting, as several people and myself have done exactly that, which then allows the IDE to open.
But when we go to the board manager to add new boards, a red error (java.lang.nullpointerexception ) at the bottom of board manager appears, and not all boards for installing appear in the  board manager.
Are you 100 percent sure you went into the board manager menu?

[38 more replies]

closed 2020-06-12 by matthijskooijman