arduino/Arduino #10333: arduino stuck at initializing package

kikiloaw, 2020-06-09:
here the problem with arduino
it was used many time but suddenly the arduino wont run now
in debug it says:
Loading configuration...
```
Initializing packages...
java.lang.NullPointerException
        at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:124)
        at processing.app.BaseNoGui.initPackages(BaseNoGui.java:484)
        at processing.app.Base.<init>(Base.java:268)
        at processing.app.Base.main(Base.java:151)
```

i tried to downgrade the version since it was stuck using latest version. after the isntalltion of arduino 1.8.9 it opens normaly. but when close and open again. the problem back again till now.

bmitov, 2020-06-09:
I have the same problem. It started this morning. It happens on all of my computers. I even tested a new Arduino IDE install in clean Virtual machine. The same problem.
The problem is due to bug in the board manager.
To reproduce the problem:

1. Make sure you are connected to internet.
2. Start the Arduino IDE
3. Open the Board manager. You will see an Java Null pointer error reported after it is opened.
4. Close the Board Manager
5. Close the Arduino IDE
6. Try to start the Arduino IDE. It will show the splash and then shut down.

If you start with arduino_debug.exe you will see the following error reported:

Loading configuration...
Initializing packages...
java.lang.NullPointerException
        at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:134)
        at processing.app.BaseNoGui.initPackages(BaseNoGui.java:504)
        at processing.app.Base.<init>(Base.java:273)
        at processing.app.Base.main(Base.java:150)

It appears that tjis is crash caused by some recent update to the board manager indexes etc.

877dev, 2020-06-09:
Same issue here, tried different machines and builds, Windows/portable.

AngryArduino, 2020-06-09:
Same error also with the libraries manager.

rtnate, 2020-06-09:
Having the same issue.   Appears to be caused by the updated package_index.json being downloaded.

amorphis984, 2020-06-09:
I have a new computer and I'm having the same problem with Arduino IDE. I quickly get the message "Initializing Package" but never starts. Does anyone know of a fix?

ptrooms, 2020-06-09:
Have the same as reported.

Removing/clearing the *.json allow restrat of Arduino but as soon one tries to use BoardManager, the problem returns (inability to start Arduino due to null-exception).

Further testing, show that (my problem) may relate to the signature file "package_index.json.sig", as soon as I remove that file and/or  change the original contents of "package_index.json" (which forces an unmatched signature), I can restart Arduino with the per BoardManager newly updated json file.

Problem remains of course that whenever I try to use BoardManager, which will renew this signature file and re-iterate the start of the problem (java null-exception after/due to Boardmanager).

I also compared "package_index.json" with te preserved old one (dated january 2020) and found that new file now of  contains the followe extra packages:  arduino-beta, Arduino megaAVR Boards 1.8.6 (in addition tot 1.8.5),  Arduino nRF528x Boards (Mbed OS), avr-gcc 7.3.0-atmel3.6.1-arduino7 (in addition to 7.3.0-atmel3.6.1-arduino5.

rtnate, 2020-06-09:
arduino-beta might be the source of the issue.  It lists tool dependencies openocd  version 0.10.0-arduino12 and dfu-util version 0.9.0-arduino2.  Neither of these tools are defined in the package_index.json.  Arduino version 1.9-beta spits out an index error without crashing.  I am up and running (for today anyway) on 1.9-beta. (Windows 10 Pro).

GregorioDiStefano, 2020-06-09:
Same issue here. Unable to start `arduino`
Tried first: arduino-1.8.10, then upgraded to arduino-1.8.12, and same exact issue.

GregorioDiStefano, 2020-06-09:
Beta 1.9 works.

ptrooms, 2020-06-09:
Nightly (03jun20) works also and do report missin packages.
```
Index error: could not find referenced tool name=openocd version=0.10.0-arduino12 packager=arduino
Index error: could not find referenced tool name=dfu-util version=0.9.0-arduino2 packager=arduino

```
@rtnate  likely the production release  cannot cope with missing packages.

jbreu, 2020-06-09:
I have exactly the same issue. I just started using Ardunio IDE today and could use it for some hours nicely. Then I needed to restart the system and afterwards it was not starting. By deleting the cached json files the IDE starts again, but as soon as I launch Board Manager it breaks with Java Exceptions.

PowerBroker2, 2020-06-09:
Using the following Python sketch, I was able to determine two possibly faulty package downloads that may be causing the problem:
```Python
import os
import json
import urllib.request
from pprint import pprint

if __name__ == '__main__':
    with open(r'C:\Users\ltber\AppData\Local\Arduino15\package_index.json', 'r') as jfile:
        contents = jfile.read()
    
    jcontents = json.loads(contents)
    
    for package in jcontents['packages']:
        for platform in package['platforms']:
            print('-' * 50)
            pprint(platform['url'])
            
            (loc, r) = urllib.request.urlretrieve(platform['url'], os.path.join(r'C:\Users\ltber\OneDrive\Desktop\Python_Stuff\source_dir', os.path.basename(platform['url'])))
            
            print(os.stat(loc).st_size)
            
            if os.stat(loc).st_size != int(platform['size']):
                print('ERROR: Incorrect download size: Expected {} bytes, but got {}'.format(platform['size'], os.stat(loc).st_size))
            
            print()
    
    print('*' * 100)
    print('Done')
```

The output of the above code shows that `http://downloads.arduino.cc/cores/Arrow-samd-1.2.0.tar.bz2` and `https://github.com/AtmelUniversityFrance/atmel-avr-xmini-boardmanagermodule/releases/download/v0.5.0/atmel-avr-xmini-boardmanagermodule-0.5.0.tar.bz2` have incorrect download sizes.

Not sure if the packages were updated and the package JSON file wasn't (or vice versa). Updating the `http://downloads.arduino.cc/cores/Arrow-samd-1.2.0.tar.bz2` file size to 36768 in the JSON file and `https://github.com/AtmelUniversityFrance/atmel-avr-xmini-boardmanagermodule/releases/download/v0.5.0/atmel-avr-xmini-boardmanagermodule-0.5.0.tar.bz2` file size to 95437 in the JSON file allowed the IDE to reopen. However, the IDE may have been allowed to open by ignoring the JSON file completely doe to the package signature becoming out of date.

How is the package sig file generated? Is it a sort of hash?

ArnoutVos, 2020-06-09:
Same here. Fresh Ubuntu 20.04 install.

[image]

PowerBroker2, 2020-06-09:
Possibly more context [here](https://forum.arduino.cc/index.php?topic=324095.msg2253545#msg2253545)

ptrooms, 2020-06-09:
I presume the signature is OK, the problem looks to be that the production release of Arduino does not like the unresolved package referenced to "openocd & dfu-util"  as referenced by "package_index.json" which is updated/checked during board-manager and (re)start.

MIKE1106, 2020-06-09:
I have the same problem with 1.8.12.  As soon as I start, it briefly shows splash screen with "Initializing" words then everything goes away.   It just started today for both my Windows 10 Home and Windows 10 Pro.
  
I downloaded and can get the 1.9 Beta to open.  I have been developing a sketch that uses the Nano 33 board.  When I open Arduino 1.9 my program loads and I get a series of messages in the output area below my sketch that read as follows:

"Index error: could not find referenced tool name=openocd version=0.10.1-arduino12 package=arduino"
"Index error: could not find referenced tool name=openocd version=0.9.0-arduino2 package=arduino"

I have the Arduino AVR Boards installed for 1.8.2

Hope someone can help?

KCole313, 2020-06-09:
Same issue here. Seems like there's an issue with the current version of the package_index.json. Delete the file and everything works until it's restored by the package manager. I'm assuming it's been updated for something in the beta build but the changes are breaking stable releases. Whatever changes were made should be reverted to restore stability until a fix is in place.

jmc-codeless, 2020-06-09:
I had the 1.8.12 Arduino IDE open and could edit scripts this afternoon.  I updated my Linux Mint (19.3 Tricia MATE Edition) with 42+ packages affected.  After that, the IDE will not launch; it says Initializing Packages... and then quits/disappears.  Uninstalled and then reinstalled Arduino - no joy.  Linux Pkg Mgr History shows a few of the installed packages.  I do not know anything about these, what they do, or if they make any difference:
arduino (2:1.0.5+dfsg2-4.1)
arduino-core (2:1.0.5+dfsg2-4.1)
avr-libc (1:2.0.0+Atmel3.6.0-1)
avrdude (6.3-4)
binutils-avr (2.26.20160125+Atmel3.6.0-1)
gcc-avr (1:5.4.0+Atmel3.6.0-1build1)
libftdi1 (0.20-4build3)
libjna-java (4.5.1-1)
libjna-jni (4.5.1-1)
librxtx-java (2.2pre2+dfsg1-1ubuntu1)

zeucxb, 2020-06-09:
Same using MacOS

mtnbkr88, 2020-06-09:
Looking forward to the package_index.json owners fixing this...

kikiloaw, 2020-06-09:
remove the ..   "package_index.json" 
this should work for you

bmitov, 2020-06-09:
> remove the .. "package_index.json"
> this should work for you
It does not. As soon as I open the Board Manager to install board support the problem starts again :-(

kikiloaw, 2020-06-09:
as of now atleast you can open arduino with built in board.
i hope they fix anytime soon

PowerBroker2, 2020-06-09:
@kikiloaw , That only allows you to open the IDE, but not fix the underlying problem

jPerotto, 2020-06-10:
https://forum.arduino.cc/index.php?topic=689307.msg4635898#msg4635898

temporary solution:

1) Disable machine's networking.
2) Copy the .../arduino15 directory to,say, arduino15.backup
3) Remove the files package_index.json and package_index.json.sig from the ..../arduino15
4) Start the IDE. It should start without network access. Go to Preferences, disable "check for updates at start". Stop the IDE.
5) Start networking
6) Start IDE. It starts.
7) Avoid any touching of the network resources! The board_manager or library_manager buggers the package_index again and again.

But add firewall exit rules from arduino.exe, arduino_debug.exe, arduino-builder.exe and arduino-cli.exe.

Hoototo, 2020-06-10:
I met the same problem, so I spent all day trying to solve it. I hope engineers can fix it soon.
Finally, I use Arduino nightly version to solve it temporarily. I also put related hardware and tools to the path of the nightly version. So Arduino can recognize the hardware.

bryanarg, 2020-06-10:
tengo el mismo problema, no se si es es algo malo con mi IDE o es algo a nivel general, he probado subir datos con el editor web y ahi no tengo problema

Gustavomurta, 2020-06-10:
The file is corrupted = package_index at
C:\Users\XXX\AppData\Local\Arduino15
You can delete this file to run IDE, but after updating of Board Manger it will crash again.
We must wait for the solution from Arduino.
This file is downloaded from
https://downloads.arduino.cc/packages/package_index.json

ptrooms, 2020-06-10:
If one need to continue developing, installing the nightly version is (for me ) good to go.

However, there might be other issues as one like/need to develop using the old versions for existing products.  In my case, I simply - as I used Linux -  installed (merely copy placed aside) the Arduino untarred version in a different location and by this keeping everything else the same.
The only thing required is to activate BoardManager in order to ensure integrity.

We have to wait, as said multiple times, until  the package_index file is fixed by the owner... whoever that may (going to) be.

ArnoutVos, 2020-06-10:
This json seems to be the problem, but the IDE should never fail with a NullPointerException. 
This thread was hard to find... normal users will think their machine is broken (as I did too).

cmaglie, 2020-06-10:
We had a problem on our package_index.json that caused the error and prevented the IDE to start.
Now the package_index has been fixed, but to recover a working installation you must delete once again the package_index.json with the cache folder and restart the IDE as usual.

Windows:
* remove file C:\Users\YourUsername\AppData\Local\Arduino15\package_index.json
* remove folder C:\Users\YourUsername\AppData\Local\Arduino15\cache

Mac:
* remove file /Users/YourUsername/Library/Arduino15/package_index.json
* remove folder /Users/YourUsername/Library/Arduino15/cache

Linux:
* remove file /home/YourUsername/.arduino15/package_index.json
* remove folder /home/YourUsername/.arduino15/cache

Please note that the problem that prevents the IDE to start has already been fixed and it's ready for the next release, that's the reason why the Nightly/Beta Builds are not affected, and also the reason why we did not detect this problem earlier, sorry about that!

closed 2020-06-10 by cmaglie