arduino/Arduino #10335: Board manager error java.lang.NullPointException

nicholasb2101, 2020-06-09:
I am unable to properly add new boards under the board manager which states java.lang.NullPointException in red along the bottom of the window and outputs this:

`Exception in thread "AWT-EventQueue-0" java.lang.NullPointerException
	at cc.arduino.contributions.packages.ui.ContributedPlatformTableCellJPanel.update(ContributedPlatformTableCellJPanel.java:222)
	at cc.arduino.contributions.packages.ui.ContributedPlatformTableCellRenderer.getTableCellRendererComponent(ContributedPlatformTableCellRenderer.java:47)
	at javax.swing.JTable.prepareRenderer(JTable.java:5729)
	at javax.swing.plaf.basic.BasicTableUI.paintCell(BasicTableUI.java:2114)
	at javax.swing.plaf.basic.BasicTableUI.paintCells(BasicTableUI.java:2016)
	at javax.swing.plaf.basic.BasicTableUI.paint(BasicTableUI.java:1812)
	at javax.swing.plaf.ComponentUI.update(ComponentUI.java:161)
	at javax.swing.JComponent.paintComponent(JComponent.java:780)
	at javax.swing.JComponent.paint(JComponent.java:1056)
	at javax.swing.JComponent.paintChildren(JComponent.java:889)
	at javax.swing.JComponent.paint(JComponent.java:1065)
	at javax.swing.JViewport.paint(JViewport.java:728)
	at javax.swing.JComponent.paintChildren(JComponent.java:889)
	at javax.swing.JComponent.paint(JComponent.java:1065)
	at javax.swing.JComponent.paintChildren(JComponent.java:889)
	at javax.swing.JComponent.paint(JComponent.java:1065)
	at javax.swing.JComponent.paintChildren(JComponent.java:889)
	at javax.swing.JComponent.paint(JComponent.java:1065)
	at javax.swing.JLayeredPane.paint(JLayeredPane.java:586)
	at javax.swing.JComponent.paintChildren(JComponent.java:889)
	at javax.swing.JComponent.paintToOffscreen(JComponent.java:5217)
	at javax.swing.RepaintManager$PaintManager.paintDoubleBuffered(RepaintManager.java:1579)
	at javax.swing.RepaintManager$PaintManager.paint(RepaintManager.java:1502)
	at javax.swing.RepaintManager.paint(RepaintManager.java:1272)
	at javax.swing.JComponent.paint(JComponent.java:1042)
	at java.awt.GraphicsCallback$PaintCallback.run(GraphicsCallback.java:39)
	at sun.awt.SunGraphicsCallback.runOneComponent(SunGraphicsCallback.java:79)
	at sun.awt.SunGraphicsCallback.runComponents(SunGraphicsCallback.java:116)
	at java.awt.Container.paint(Container.java:1978)
	at java.awt.Window.paint(Window.java:3906)
	at javax.swing.RepaintManager$4.run(RepaintManager.java:842)
	at javax.swing.RepaintManager$4.run(RepaintManager.java:814)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at javax.swing.RepaintManager.paintDirtyRegions(RepaintManager.java:814)
	at javax.swing.RepaintManager.paintDirtyRegions(RepaintManager.java:789)
	at javax.swing.RepaintManager.prePaintDirtyRegions(RepaintManager.java:738)
	at javax.swing.RepaintManager.access$1200(RepaintManager.java:64)
	at javax.swing.RepaintManager$ProcessingRunnable.run(RepaintManager.java:1732)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:109)
	at java.awt.WaitDispatchSupport$2.run(WaitDispatchSupport.java:190)
	at java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:235)
	at java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:233)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.awt.WaitDispatchSupport.enter(WaitDispatchSupport.java:233)
	at java.awt.Dialog.show(Dialog.java:1084)
	at java.awt.Component.show(Component.java:1671)
	at java.awt.Component.setVisible(Component.java:1623)
	at java.awt.Window.setVisible(Window.java:1014)
	at java.awt.Dialog.setVisible(Dialog.java:1005)
	at processing.app.Base.openBoardsManager(Base.java:1411)
	at processing.app.Base$10.actionPerformed(Base.java:1438)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.AbstractButton.doClick(AbstractButton.java:376)
[... 153 lines cut ...]
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4904)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4535)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4476)
	at java.awt.Container.dispatchEventImpl(Container.java:2283)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:760)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:84)
	at java.awt.EventQueue$4.run(EventQueue.java:733)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:730)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
`

aidzer0, 2020-06-09:
I am getting the same issues on windows 10 with 1.8.12. Removed and re-installed both Arduino and Java with no success, tried with and without adding boards. This happens when Board Manager is opened.

nicholasb2101, 2020-06-09:
Sounds like this is an error on the board's manager package updating when loading.

KCole313, 2020-06-10:
Same root cause as #10331, #10332, and #10333.

matricce, 2020-06-10:
test this fork https://github.com/ricardojlrufino/Arduino/branches
working here on W10

TheWebMachine, 2020-06-10:
While it was working perfectly on TWO systems as recently as 13 hours ago, neither are working now. Boards Manager AND Library Manager give the same exact error after trying to obtain updates. A bad json file released today on arduino.cc, perhaps?

mjs513, 2020-06-10:
Think I have a related issue where the IDE fails to start and dies on loading packages.  I just posted on the forum: https://forum.arduino.cc/index.php?topic=671157.new#new

QUOTE:
Well here is another of the same issue.

I deleted the Arduino15 the IDE 1.8.12 would then start and when I closed it and tried to reopen the IDE would not start.  Seems to be dying at loading package manager.  This just started today.  

Running Arduino_debug I get the following:

F:\Program Files (x86)\Arduino>arduino_debug
Set log4j store directory C:\Users\Merli\AppData\Local\Arduino15
Loading configuration...
Initializing packages...
java.lang.NullPointerException
        at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:134)
        at processing.app.BaseNoGui.initPackages(BaseNoGui.java:504)
        at processing.app.Base.<init>(Base.java:273)
        at processing.app.Base.main(Base.java:150)

I am running on a Windows 10x64 version 1909 build 18363.836.

Gustavomurta, 2020-06-10:
In my case 👍 
C:\Program Files (x86)\Arduino>arduino_debug
Carregando configuração...
Inicializando pacotes...
java.lang.NullPointerException
        at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:124)
        at processing.app.BaseNoGui.initPackages(BaseNoGui.java:484)
        at processing.app.Base.<init>(Base.java:268)
        at processing.app.Base.main(Base.java:151)

C:\Program Files (x86)\Arduino>

chefer, 2020-06-10:
Gustavomurta, I have the same history and same issue !

Gustavomurta, 2020-06-10:
I think is some bad package. I'm reading application log to see if something is wrong.
C:\Users\XXXX\AppData\Local\Arduino15\logs

Gustavomurta, 2020-06-10:
at processing.app.BaseNoGui.initPackages(BaseNoGui.java:484)

https://github.com/arduino/Arduino/blob/07b3b9036a22ea389ee936c10503d5e97ab26730/arduino-core/src/processing/app/BaseNoGui.java

Gustavomurta, 2020-06-10:
I think I found the problem!
The file is corrupted = package_index at 
C:\Users\XXX\AppData\Local\Arduino15
I rename this file and Arduino now running. 
But I lose all packages.

chefer, 2020-06-10:
Apparently it works. Night construction.

https://www.arduino.cc/download.php?f=/arduino-nightly-windows.zip

Gustavomurta, 2020-06-10:
Cheffer, please add ESP32 Package an test the IDE.

chefer, 2020-06-10:
Gustavomurta, eu adicionei ESP8266 e funcionou! ;)

Gustavomurta, 2020-06-10:
Please, update the Boards. Then the IDE will crash.

Gustavomurta, 2020-06-10:
Downloaded package index URL=[http://arduino.esp8266.com/stable/package_esp8266com_index.json, https://dl.espressif.com/dl/package_esp32_index.json, https://downloads.arduino.cc/packages/package_index.json]

Gustavomurta, 2020-06-10:
The file is corrupted = package_index at
C:\Users\XXX\AppData\Local\Arduino15
You can delete this file to run IDE, but after updating of Board Manger it will crash again.
We must wait for the sollution from Arduino.
This file is downloaded from
https://downloads.arduino.cc/packages/package_index.json

vk1ajm, 2020-06-10:
I agree 100% with the above post as I am experiencing the same same issues  after trying to use board manager.  Board Manager fails with index issues with the base packages.

valesavenko, 2020-06-10:
+ 1 here on my end. Spent hours reinstalling, fixing, hacking. Glad found this thread to confirm my thinking.

syd1986, 2020-06-10:
Same here. 
Short Term Solution: The hourly build version works fine:
https://www.arduino.cc/download.php?f=/arduino-nightly-windows.zip

facchinm, 2020-06-10:
We had a problem on our package_index.json that caused the error and prevented the IDE to start.
Now the package_index has been fixed, but to recover a working installation you must delete once again the package_index.json with the cache folder and restart the IDE as usual.

Windows:
* remove file C:\Users\YourUsername\AppData\Local\Arduino15\package_index.json
* remove folder C:\Users\YourUsername\AppData\Local\Arduino15\cache

Mac:
* remove file /Users/YourUsername/Library/Arduino15/package_index.json
* remove folder /Users/YourUsername/Library/Arduino15/cache

Linux:
* remove file /home/YourUsername/.arduino15/package_index.json
* remove folder /home/YourUsername/.arduino15/cache

Please note that the problem that prevents the IDE to start has already been fixed and it's ready for the next release, that's the reason why the Nightly/Beta Builds are not affected, and also the reason why we did not detect this problem earlier, sorry about that!

closed 2020-12-24 by per1234

Barrut827, 2021-04-26:
cant find these on ubuntu 20.04:

Linux:

    remove file /home/YourUsername/.arduino15/package_index.json
    remove folder /home/YourUsername/.arduino15/cache

any clue where to look?

TheWebMachine, 2021-04-26:
You likely installed the snap version. In this case, look here:

`~/snap/arduino/current/.arduino15/`
(~ is your home directory; usually /home/{username}/)

Barrut827, 2021-04-26:
> You likely installed the snap version. In this case, look here:
> 
> `~/snap/arduino/current/.arduino15/`
> (~ is your home directory; usually /home/{username}/)

well the .arduino15 part i dont have. only a Arduino folder inside this current folder.