arduino/Arduino #10336: NPE in BoardsManager if Arduino15 directory removed, crash at start if exists

afunix, 2020-06-10:
Arduino IDE 1.8.12 on MacOS 10.15.5.

When I launch IDE and open Board Manager I get an NPE:
```
java.lang.NullPointerException
java.lang.RuntimeException: java.lang.NullPointerException
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$1(ContributionManagerUI.java:150)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.NullPointerException
	at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:134)
	at processing.app.BaseNoGui.initPackages(BaseNoGui.java:483)
	at processing.app.Base$9.onIndexesUpdated(Base.java:1400)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$1(ContributionManagerUI.java:148)
	... 1 more
```
Attached log ([arduino-ide--board-manager.log](https://github.com/arduino/Arduino/files/4755509/arduino-ide--board-manager.log)) since the IDE launch.

Noticed HTTP 206 Partial Content in log, not sure if it is related
```
Request complete URL="https://downloads.arduino.cc/packages/package_index.json.sig", method=GET, response code=206, request id=7724CA897B9D4049, headers={null=[HTTP/1.1 206 Partial Content], Server=[cloudflare],
```

Later if IDE is restarted it crashes at start with similar NPE:
```
java.lang.NullPointerException
	at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:134)
	at processing.app.BaseNoGui.initPackages(BaseNoGui.java:483)
	at processing.app.Base.<init>(Base.java:273)
	at processing.app.Base.main(Base.java:150)
```
Attached log ([arduino-ide--startup.log](https://github.com/arduino/Arduino/files/4755524/arduino-ide--startup.log)) with crash at startup.

Tried to remove package_index.json and package_index.json.sig files.
Tried to remove ~/Library/Arduino15 directory.
Nothing helps.

Attaching package_index.json and package_index.json.sig ([package_index.json.zip](https://github.com/arduino/Arduino/files/4755536/package_index.json.zip)) just in case.

bernardo-amaral, 2020-06-10:
I put this file on read only mode "C:\Users\XXXXX\AppData\Local\Arduino15\package_index.json" not i can use the IDE.. 
If you need download esp board, rename "package_esp8266com_index.json" to "package_index.json", and set this file to readonly.

Works for me...

afunix, 2020-06-10:
I've installed Arduino IDE 1.9.0 Beta and it complained when started with:
[image]

I indeed found directory "hd44780 2" in ~/Documents/Arduino/libraries.
I guess it was created during the library update.
Removed ~/Library/Arduino15/staging/libraries/hd44780* and ~/Documents/Arduino/libraries/hd44780*, restarted IDE and reinstalled the package.

1.9.0 Beta seems to work fine now, however released 1.8.12 still crashes at start up.

Gustavomurta, 2020-06-10:
My IDE 1.8.12 not open too! And of some friends too. 
I think was some bad update with bug!

afunix, 2020-06-10:
FYI, I've tried to downgrade to 1.8.11 and it fails the same way.
I believe there is something in packages files out there in the cloud, which makes IDE to crash trying to refresh the package list.

Gustavomurta, 2020-06-10:
I think is some bad package. I'm reading application log to see if something is wrong.
C:\Users\XXXX\AppData\Local\Arduino15\logs

alanzoppa, 2020-06-10:
Running an otherwise fresh Lubuntu VM on an OSX host, having installed 1.8.12 from the 64-bit Linux binary:

After adding the ESP8266 etc. board URL in Arduino prefs, I consistently get the exception that @afunix describes. It goes away if I remove ~/.arduino15, but that's no good.

Gustavomurta, 2020-06-10:
The file is corrupted = **package_index** at
C:\Users\XXX\AppData\Local\Arduino15
You can delete this file to run IDE, but after update of Board Manger it will crash again.
We must wait the sollution from Arduino. 
This file is downloaded from 
https://downloads.arduino.cc/packages/package_index.json

facchinm, 2020-06-10:
We had a problem on our package_index.json that caused the error and prevented the IDE to start.
Now the package_index has been fixed, but to recover a working installation you must delete once again the package_index.json with the cache folder and restart the IDE as usual.

Windows:
* remove file C:\Users\YourUsername\AppData\Local\Arduino15\package_index.json
* remove folder C:\Users\YourUsername\AppData\Local\Arduino15\cache

Mac:
* remove file /Users/YourUsername/Library/Arduino15/package_index.json
* remove folder /Users/YourUsername/Library/Arduino15/cache

Linux:
* remove file /home/YourUsername/.arduino15/package_index.json
* remove folder /home/YourUsername/.arduino15/cache

Please note that the problem that prevents the IDE to start has already been fixed and it's ready for the next release, that's the reason why the Nightly/Beta Builds are not affected, and also the reason why we did not detect this problem earlier, sorry about that!

closed 2020-06-10 by facchinm