arduino/Arduino #10337: Preprocessor directives error

geotrieu, 2020-06-10:
The following #define give me errors in my program:

`#define ENVIRONMENT "dev"`

returns the error 
```
error: token ""dev"" is not valid in preprocessor expressions
 #define ENVIRONMENT "dev"
```

matthijskooijman, 2020-06-10:
I suspect the problem is not in the definition of the macro, but in its usage.

Can you post a full (but short) sketch that shows the problem along with the IDE output? What board and IDE version are you using?

geotrieu, 2020-06-11:
Here is a sketch that errors out on me:
```
#define ENV "dev"

#if ENV == "prod"
  //Do this
#else
  //Do that
#endif

void setup() {
  // put your setup code here, to run once:
}
void loop() {
  // put your main code here, to run repeatedly:
}
```
I am using Arduino 1.8.12, and succeeded in replicating the problem in the latest nightly build.
I find it weird because in my actual program, a couple lines below has
`#define MQTT_SERVER "192.168.0.3"`
and that works just fine.

I am also using the ESP8266 board, however the issue persists even when the Arduino Uno is selected.

matthijskooijman, 2020-06-11:
Yup, as I suspected, the problem is in how you use the macro. In the preprocessor, string comparison is AFAIU simply not allowed/implemented. The preprocessor can use strings (or pretty much any value) inside macros and replace them pretty much everywhere, but as soon as the preprocessor has to actually interpret the values (e.g. in `#if`), it does not support strings. So you cannot do `#if "dev" == "prod"` (which is your example after macro expansion).

See also https://stackoverflow.com/questions/2335888/how-to-compare-strings-in-c-conditional-preprocessor-directives

Since this seems to be working as intended (and documented, though maybe not explicitly by Arduino), I'm going to close this issue. Feel free to add further comments if needed.

closed 2020-06-11 by matthijskooijman