arduino/Arduino #10339: [Library Manager] Add Library

RockoonTechnologies, 2020-06-10:
please add this library: https://github.com/RockoonTechnologies/PadComLib

per1234, 2020-06-10:
Hi @RockoonTechnologies your library does not currently meet the requirements for addition to the Library Manager index.

You are not allowed to have the library in a .zip file. Please put the library files directly into the repository and delete the .zip file. GitHub provides automatically generated .zip files so there is no need for you to put the library in a .zip file yourself (and in fact doing that would have a very harmful effect on version control and collaboration).

Your library.properties is also missing the required `url` field.

After resolving these issues, please make a new release/tag or redo the existing one, then comment here so we'll know we can proceed.

More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

per1234, 2020-06-11:
It's still not compliant. The library must be in the root of the repository, not in a subfolder. Please fix the structure of the repository and then either make a new release or else redo the existing release and then comment here.

RockoonTechnologies, 2020-06-11:
https://github.com/RockoonTechnologies/PadComLib

per1234, 2020-06-11:
Still the same problem, only worse now because you moved library.properties under the src subfolder.

This is the current structure of your repository:
```
PadComLib
|_ README.md
|_ PadComLib
    |_ src
        |_ PadComLib.cpp
        |_ PadComLib.h
        |_ library.properties
```
Note that the library is under the PadComLib subfolder of the PadComLib repository. That is not allowed.

This is the structure required for your library to be added to Library Manager:
```
PadComLib
|_ library.properties
|_ README.md
|_ src
    |_ PadComLib.cpp
    |_ PadComLib.h
```
You  can find more information in the [Library Manager FAQ](https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ) and the [Arduino Library Specification](https://arduino.github.io/arduino-cli/latest/library-specification/).

RockoonTechnologies, 2020-06-12:
Ok, I updated it, im very sorry for the trouble I caused, I am new to libs

https://github.com/RockoonTechnologies/PadComLib

per1234, 2020-06-12:
The repository looks good now, but the Library Manager indexer works from tags/releases, and at the time of your last release, the library was still not compliant. So now you only need to create a new release and you'll be ready to go.

RockoonTechnologies, 2020-06-12:
Okie dokie, I added a new release!

cmaglie, 2020-06-22:
Done.

closed 2020-06-22 by cmaglie