arduino/Arduino #10340: Arduino Environment exists before it's initialized

leonbrag, 2020-06-10:
Version: 1.8.12

:\Program Files (x86)\Arduino>arduino_debug.exe
Set log4j store directory C:\Users\leonb\AppData\Local\Arduino15
Loading configuration...
Initializing packages...
java.lang.NullPointerException
        at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:134)
        at processing.app.BaseNoGui.initPackages(BaseNoGui.java:483)
        at processing.app.Base.<init>(Base.java:273)
        at processing.app.Base.main(Base.java:150)

Seems like this issue: https://github.com/arduino/Arduino/issues/3248

It worked for weeks. Suddenly it stopped.

imrehorvath, 2020-06-10:
+1

Opening the Board Manager and having the packages updated seems to trigger this issue. After this, the problem seems to be permanent.

glkada, 2020-06-10:
My Arduino IDE too shows the same problem. It crashes as soon as I launch it. 
[image]

leonbrag, 2020-06-10:
Un-installed, Re-install did NOT resolve the problem.

As a fix I have installed Windows Store app. Store app works.

cmaglie, 2020-06-10:
We had a problem on our package_index.json that caused the error and prevented the IDE to start.
Now the package_index has been fixed, but to recover a working installation you must delete once again the package_index.json with the cache folder and restart the IDE as usual.

Windows:
* remove file C:\Users\YourUsername\AppData\Local\Arduino15\package_index.json
* remove folder C:\Users\YourUsername\AppData\Local\Arduino15\cache

Mac:
* remove file /Users/YourUsername/Library/Arduino15/package_index.json
* remove folder /Users/YourUsername/Library/Arduino15/cache

Linux:
* remove file /home/YourUsername/.arduino15/package_index.json
* remove folder /home/YourUsername/.arduino15/cache

Please note that the problem that prevents the IDE to start has already been fixed and it's ready for the next release, that's the reason why the Nightly/Beta Builds are not affected, and also the reason why we did not detect this problem earlier, sorry about that!

closed 2020-06-10 by cmaglie