arduino/Arduino #10341: Arduino IDE can not open ！Something must be wrong ！

xp187oo, 2020-06-10:
java.lang.NullPointerException
        at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:134)
        at processing.app.BaseNoGui.initPackages(BaseNoGui.java:483)
        at processing.app.Base.<init>(Base.java:273)
        at processing.app.Base.main(Base.java:150)

AkhilGKrishnan, 2020-06-10:
@xp187oo  I am facing the same issue. when i open board manager

hmeijdam, 2020-06-10:
If you look at the forum  https://forum.arduino.cc/index.php?board=2.0  you will see that everyone has this problem. One of the files that the "boards manager" is downloading must cause a data corruption. I have it in the older IDE versions also, so not just 1.8.12

jandrassy, 2020-06-10:
it looks like Arduino's package_index.json is the corrupted one. no way around

imrehorvath, 2020-06-10:
Duplicate of https://github.com/arduino/Arduino/issues/10340

jandrassy, 2020-06-10:
problem is a new section `arduino-beta` for Portenta H7

cmaglie, 2020-06-10:
We had a problem on our package_index.json that caused the error and prevented the IDE to start.
Now the package_index has been fixed, but to recover a working installation you must delete once again the package_index.json with the cache folder and restart the IDE as usual.

Windows:
* remove file C:\Users\YourUsername\AppData\Local\Arduino15\package_index.json
* remove folder C:\Users\YourUsername\AppData\Local\Arduino15\cache

Mac:
* remove file /Users/YourUsername/Library/Arduino15/package_index.json
* remove folder /Users/YourUsername/Library/Arduino15/cache

Linux:
* remove file /home/YourUsername/.arduino15/package_index.json
* remove folder /home/YourUsername/.arduino15/cache

Please note that the problem that prevents the IDE to start has already been fixed and it's ready for the next release, that's the reason why the Nightly/Beta Builds are not affected, and also the reason why we did not detect this problem earlier, sorry about that!

closed 2020-06-10 by cmaglie

Hamid11300, 2020-06-16:
> We had a problem on our package_index.json that caused the error and prevented the IDE to start.
> Now the package_index has been fixed, but to recover a working installation you must delete once [...]

Big Like

xp187oo, 2020-06-16:
Thank you for replay，man.   I had empty C:\Users\YourUsername\AppData\Local\Arduino15\   and  it works .