arduino/Arduino #10342: Arduino IDE doesn't start

Nitheshnithu121, 2020-06-10:
I have a problem when i start Arduino IDE it starts, then initializes packages and just vanishes......
The output on running debugger is as follows:
C:\Program Files (x86)\Arduino>arduino_debug.exe
Set log4j store directory C:\Users\user\AppData\Local\Arduino15
Loading configuration...
Initializing packages...
java.lang.NullPointerException
at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:134)
at processing.app.BaseNoGui.initPackages(BaseNoGui.java:483)
at processing.app.Base.(Base.java:273)
at processing.app.Base.main(Base.java:150)

Cloud any one please help me solve this problem

imrehorvath, 2020-06-10:
Duplicate of https://github.com/arduino/Arduino/issues/10340
and https://github.com/arduino/Arduino/issues/10341

Nitheshnithu121, 2020-06-10:
> Duplicate of #10340
> and #10341

i am not able to solve this issue cloud you please help me to solve this problem

195cn, 2020-06-10:
I have this promblem too.
After I delete package_index.json in C:\Users\xxx\AppData\Local\Arduino15 , IDE is able to use.
But when I open board manager, NullPointerException problem appear again.
Then IDE crash at read package list , after close it.
----------------
I already reset my win10,java jdk.
And 1.8.10 , 1.8.12 appear this problem too.
----------------
 
[image]

Nitheshnithu121, 2020-06-10:
How can solve this problem bro

cmaglie, 2020-06-10:
We had a problem on our package_index.json that caused the error and prevented the IDE to start.
Now the package_index has been fixed, but to recover a working installation you must delete once again the package_index.json with the cache folder and restart the IDE as usual.

Windows:
* remove file C:\Users\YourUsername\AppData\Local\Arduino15\package_index.json
* remove folder C:\Users\YourUsername\AppData\Local\Arduino15\cache

Mac:
* remove file /Users/YourUsername/Library/Arduino15/package_index.json
* remove folder /Users/YourUsername/Library/Arduino15/cache

Linux:
* remove file /home/YourUsername/.arduino15/package_index.json
* remove folder /home/YourUsername/.arduino15/cache

Please note that the problem that prevents the IDE to start has already been fixed and it's ready for the next release, that's the reason why the Nightly/Beta Builds are not affected, and also the reason why we did not detect this problem earlier, sorry about that!

Nitheshnithu121, 2020-06-10:
I am not finding package_index.json  file and cache inside Arduino folder

cmaglie, 2020-06-10:
@Nitheshnithu121 the folder is called `Arduino15` you can find it in:

C:\Users\YourUsername\AppData\Local\Arduino15\

Nitheshnithu121, 2020-06-10:
Thank you so much brother, its working

closed 2020-06-10 by Nitheshnithu121