arduino/Arduino #10345: Arduino IDE fails to start (with solution)

matthijskooijman, 2020-06-10:
Due to a server-side issue yesterday (June 10th), the Arduino IDE failed to start due to downloading corrupt board files. The server-side issue has since been fixed, but if you ran into this problem, some manual action is required to make the IDE start again.

Elsewhere, @cmaglie wrote the following message:

---
We had a problem on our package_index.json that caused the error and prevented the IDE to start.
Now the package_index has been fixed, but to recover a working installation you must delete once again the package_index.json with the cache folder and restart the IDE as usual.

Windows:

    remove file C:\Users\YourUsername\AppData\Local\Arduino15\package_index.json
    remove folder C:\Users\YourUsername\AppData\Local\Arduino15\cache

Mac:

    remove file /Users/YourUsername/Library/Arduino15/package_index.json
    remove folder /Users/YourUsername/Library/Arduino15/cache

Linux:

    remove file /home/YourUsername/.arduino15/package_index.json
    remove folder /home/YourUsername/.arduino15/cache

Please note that the problem that prevents the IDE to start has already been fixed and it's ready for the next release, that's the reason why the Nightly/Beta Builds are not affected, and also the reason why we did not detect this problem earlier, sorry about that!

---

@cmaglie, @facchinm, I'm creating *yet another* issue about this, sow we can pin this and hopefully prevent further issues from being created about this and to let people find the solution quickly (and all of the other issues had a lot of distracting comments before the solution, so I did not want to pin any of those).

@cmaglie, would upgrading to the nightly version also fix this? Or would you then stil need to remove the corrupt files?

cmaglie, 2020-06-10:
> would upgrading to the nightly version also fix this? Or would you then stil need to remove the corrupt files?

the nightly should work without removing the corrupted files.

MIKE1106, 2020-06-11:
Thanks much!!!   I'm back in business today again.

bshambaugh, 2020-06-11:
Thanks. It worked. For me the syntax was a bit different:
rm /home/ubuntu/.arduino15/package_index.json
rm -r /home/ubuntu/.arduino15/cache/

_I made backups just in case:_
cp /home/ubuntu/.arduino15/package_index.json /home/ubuntu/.arduino15/package_index.json.backup
cp -r /home/ubuntu/.arduino15/cache/ /home/ubuntu/.arduino15/cache_backup/

cmaglie, 2020-06-17:
We've now released Arduino IDE 1.8.13 that should fix this problem without the need to manually delete anything.

After opening the "Board manager" with the IDE 1.8.13 at least once, a "normal" condition is recovered and the previous IDE releases should be working again too.

closed 2020-07-31 by per1234

RAJensen, 2020-08-28:
I have just downloaded and installed 1.8.13 from Arduino.cc on a Windows 10 Pro.
Sorry to say, but the problem persist.
First time IDE starts, next time, just splash scrren and it close.
Only way to get it start again is to delete preferences.txt in /Arduino15.
I have tried with 1.8.9 and it do the same thing.
Really wierd, as i have 1.8.9 running on Windows 10 Home edition with no trouble at all.

matthijskooijman, 2020-08-28:
@RAJensen, I suspect you're having a different issue with similar symptoms that looks the same. If you can still reproduce the problem, could you create a new issue describing your problem? You should probably also run `arduino_debug.exe` from a terminal and include any output in your issue, that might offer some insight in what the problem is.

RAJensen, 2020-08-28:
>

> @RAJensen, I suspect you're having a different issue with similar symptoms that looks the same. If you can still reproduce the problem, could you create a new issue describing your problem? You [...]

When IDE won't start, arduino_debug.exe won't either. It just to the same, start up and close.
I found a guide on the net. He used CC Cleaner, so i run it too and now IDE 1.8.9 is running.
Maybe it is something in some Windows installations that create the error?
But for now, i'm a happy camper. :-)

nicholasglazer, 2020-09-02:
For Archlinux users you should install "arduino-avr-core", this will solve the issue.

Nikakz, 2020-09-02:
> Acabei de baixar e instalar 1.8.13 do Arduino.cc em um Windows 10 Pro.
> Desculpe dizer, mas o problema persiste.
> A primeira vez que o IDE é iniciado, da próxima vez, basta abrir o scrren e fechá-lo.
> A única maneira de começar de novo é excluir o arquivo preferences.txt em / Arduino15.
> Eu tentei com 1.8.9 e ele faz a mesma coisa.
> Muito estranho, pois tenho 1.8.9 rodando no Windows 10 Home edition sem nenhum problema.

Agradeço. vc está correto, deletei a preferences.txt e a IDE abriu. Obrigada já não sabia mais o que fazer.