arduino/Arduino #10349: Arduino IDE 1.8.13 Crashes (Java: Java2D Queue Flusher) When moving between monitors

allthemoose, 2020-06-11:
Hi, 

Issue is with the nightly build. Version 1.8.13 (2020/06/03).

This is my first Bug report so I apologise if I don't get it correct. 
I noticed the Arduino IDE crashes repeatably when loading on one monitor and moved to another. I have 2 additional screens connected to Macbook pro via a USB3 hub. 
Once the IDE is loaded, I move the screen to the upper monitor and it crashes spitting an error report. 
    
```
Process:               Arduino [2404]
Path:                  /Applications/Code/Arduino.app/Contents/MacOS/Arduino
Identifier:            Arduino
Version:               nightly (1.8.13)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           Arduino [2404]
User ID:               501

Date/Time:             2020-06-11 20:25:46.292 +1200
OS Version:            Mac OS X 10.15.5 (19F101)
Report Version:        12
Anonymous UUID:        AA971289-603F-3226-F5D4-D97A892DF392

Sleep/Wake UUID:       76A25BF6-BB87-4B6D-AE43-4A254DA71219

Time Awake Since Boot: 9800 seconds
Time Since Wake:       640 seconds

System Integrity Protection: enabled

Crashed Thread:        17  Java: Java2D Queue Flusher

Exception Type:        EXC_BAD_INSTRUCTION (SIGABRT)
Exception Codes:       0x0000000000000001, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
-[NSOpenGLContext setView:] must be called from the main thread.
```

per1234, 2020-06-11:
Closing as duplicate of https://github.com/arduino/Arduino/issues/9764

Please see https://github.com/arduino/Arduino/pull/10234 for updates on the fix in progress and to assist with testing.

closed 2020-06-11 by per1234

cmaglie, 2020-06-11:
@allthemoose there is a test build here -> https://github.com/arduino/Arduino/pull/10234#issuecomment-642679207

May you try it and report on #10234 your results? thanks