arduino/Arduino #10351: My code does not work.

alexlastique, 2020-06-11:
Hello, I'm French and I received a robot from your brand for Christmas, except that the accompanying program does not work, it may have been programmed for a different version from the version I use but I do not know how to enter the code of my robot successfully.

Could you tell me what I should do.

Sorry for my English, I use google translation.

Thank you in advance.

per1234, 2020-06-11:
Hi @alexlastique. I'm sorry to hear you're having trouble with your robot. In order for us to be able to help you, we'll need more information:

We need to know exactly which robot you have. If you can post a link to the product page on the [Arduino Store](https://store.arduino.cc/), that would be ideal.

We need to know which program you're talking about. Either post a link to where you downloaded the program from, or post the full program here.

We need a detailed description of what you mean by "does not work".

alexlastique, 2020-06-11:
Sory but I did not find it on your site but I found it on Amazon:
https://www.amazon.fr/Starmood-Ultrasonic-Tracking-Moteur-Module/dp/B07QDRQZCG
With the operating instructions
coding begins on page 17:
http://www.cs.columbia.edu/~sedwards/presentations/robot-car-instructions.pdf

when i want to upload the program it puts: C: \ Users \ alexi \ OneDrive \ Documents \ Arduino \ robot_voiture \ robot_car \ robot_car.ino: In function 'void setup ()':

robot_car: 21: 1: error: a function-definition is not allowed here before '{' token

 {

 ^

robot_car: 34: 1: error: a function-definition is not allowed here before '{' token

 {

 ^

robot_car: 46: 1: error: a function-definition is not allowed here before '{' token

 {

 ^

robot_car: 57: 14: error: a function-definition is not allowed here before '{' token

 void setup () {

              ^

robot_car: 82: 13: error: a function-definition is not allowed here before '{' token

 void loop () {

             ^

exit status 1
a function-definition is not allowed here before '{' token

per1234, 2020-06-12:
Although that product is compatible with Arduino, it's neither manufactured nor sold by Arduino. Since both the hardware and software are from 3rd parties, this issue tracker is not the appropriate place to discuss any possible problems with them. This topic is more appropriate for the Arduino Forum. I'm sure we'll be able to help you with your problem over there.

Please do this:
1. Read http://forum.arduino.cc/index.php?topic=148850
2. If you haven't already done so, create a new thread in the appropriate section of the forum (http://forum.arduino.cc/), following all the rules. Be sure to post your full code, using code tags (`</>` button on the toolbar).

closed 2020-06-12 by per1234