arduino/Arduino #10354: Arduino compiler fails if file "includes.cache" exists in the build directory

wcieslik, 2020-06-12:
Currently using version 1.8.13 hourly build 2020/06/03 04:33 on Windows 10 (64 bit).

A project can only be compile once, during a session.
At the first compile, 'includes.cache' is created in the build directory as expected.

Compiling the project again fails.

If the includes.cache file is manually deleted then the compile works again and a new includes.cache is created.

Closing the IDE deletes the entire build directory, which is why closing and re-opening the IDE allows a one time compile.

The 'includes.cache' file is not write protected or hidden, etc.

wcieslik, 2020-06-12:
I have found the solution to my issue ...
Ditch 1.8.12, and 1.8.13,  I tried both ....
and use 1.8.10.

1.8.10 works fine !! :-)

matthijskooijman, 2020-06-12:
What error do you get when it's not working? Does it happen with any sketch (e.g. also the bare minimum sketch, maybe with just one include added)?

wcieslik, 2020-06-14:
Hi Mathisj.....
Yes.. it happens with any sketch.
Blink .. Bare Miniumum .... doesn't matter.

Happily using 1.8.10 now instead of .11  .12  or .13 or even 1.8.5 which was my previouse  base...

Cheers.

matthijskooijman, 2020-06-15:
What error do you get when it's not working?

wcieslik, 2020-07-05:
Damn it.. I have been forced into upgrading back up to 1.8.12 again, and now face the same old problem with the "includes.cache" file...

Verifying the BLINK codes works 1st time after starting the IDE.
verifying again, (no changes to anything) results in the output pasted below ...

If I manually delete the includes.cache file, then verify works again.. one time only.

Fairly sure this is some sort of permissions/domain or local security policy or even a firewall thing, but can't find anything to indicate what the actual issue is with the IDE and this file ...

ALL 6 lines of the output from the second verify ...
the first two lines show the compiler and build options
the second two lines show the board and the core
the fifth line is simply a status message
the sixth line is the error message that simply states "Error compiling for board Arduino Nano."

Which really means "error compiling because I can't overwrite includes.cache"

As mentioned before... this doesn't happen if I use 1.8.10, and seems to start from 1.8.11 onwards ... as is specific to my laptop and probably.. security settings ...

Cheers,
Wit C.

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\e156994\Documents\Arduino\libraries -fqbn=arduino:avr:nano:cpu=atmega328old -vid-pid=1A86_7523 -ide-version=10812 -build-path C:\Users\e156994\AppData\Local\Temp\arduino_build_990350 -warnings=none -build-cache C:\Users\e156994\AppData\Local\Temp\arduino_cache_802750 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\e156994\Documents\Arduino\Projects\MyBlink\Blink\Blink.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\e156994\Documents\Arduino\libraries -fqbn=arduino:avr:nano:cpu=atmega328old -vid-pid=1A86_7523 -ide-version=10812 -build-path C:\Users\e156994\AppData\Local\Temp\arduino_build_990350 -warnings=none -build-cache C:\Users\e156994\AppData\Local\Temp\arduino_cache_802750 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\e156994\Documents\Arduino\Projects\MyBlink\Blink\Blink.ino
Using board 'nano' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Using core 'arduino' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Detecting libraries used...
Error compiling for board Arduino Nano.

wcieslik, 2020-07-05:
Sorry, didn't mean to close...

matthijskooijman, 2020-07-05:
> this doesn't happen if I use 1.8.10, and seems to start from 1.8.11 onwards

Ah, that's useful info, knowing that it started with 1.8.11 can help narrowing down the issue.

The output you provided is very weird, since it seems arduino-builder just stops with an error code, without printing any error message.

Looking at the changes to arduino-builder between IDE 1.8.10 (arduino-builder 1.4.7) and IDE 1.8.11 (arduino-builder 1.5.1), I find it includes the following changes to arduino-cli: https://github.com/arduino/arduino-cli/compare/b3db7a6400cf...d3c1d5031178 Of those, https://github.com/arduino/arduino-cli/commit/cc737536689a06ab5404529fdf73ff86821be708 seems to be the only one related to include caching, so that would be a likely suspect to cause this problem.

I don't directly see how that commit would cause this, but I would like to confirm that this commit is indeed problematic. @wcieslik could you:
 - Install 1.8.11 and confirm the problem *does* occur.
 - Replace `C:\Program Files (x86)\Arduino\arduino-builder.exe` with [this version of `arduino-builder.exe`](https://github.com/arduino/Arduino/files/4875482/arduino-builder.zip) (you'll have to unzip it, Github does not allow attaching .exe files directly) and see if the problem still occurs.

The arduino-builder version I've attached is 1.5.1 (like contained in IDE 1.8.11), but with commit cc737536689a06ab5404529fdf73ff86821be708 reverted.

wcieslik, 2020-09-18:
Apologies for leaving this so long...
I re-confirmed the version 1.8.10 works for me w.r.t. the compliler/builder, and ver 1.8.11 does not.
Same issue with the latest 1.8.13  as well.

Following you suggestion I renamed arduino-builder.exe and copied in the zipped one you sent me.
This seems to work fine !

Many thanks for your help with this...

matthijskooijman, 2020-09-18:
Thanks for the followup and additional details. So the commit I mentioned above is somehow responsible, but I'm quite unsure how exactly... No time to look deeper right now, but I'll probably need you to do some more testing in the future. I'll get back to you.

@cmaglie, any ideas?

wcieslik, 2020-09-22:
Thanks for looking into this ...

My personal theory (vague as it is ) is along the lines of :
[1] Corp Security policy on my windows laptop, that allows the earlier
version of builder to access the includes.cache
[2] Some form of IE setting or rule (again Corp Policy etc) that 'protects'
anything ending in 'cache' from being modified
or some combination of the above two ...

Cheers...

miweber67, 2020-12-25:
I am also having this issue... FWIW, it wasn't always the case, but it appears to have started when I started changing code in libraries in my documents\Arduino\libraries\... folder. Before that, I never had an issue. Don't know if that's a red herring or not, but... just another data point.

I ran arduino_debug, but got no additional information between "Detecting libraries used" and the error message.

sztimi, 2021-01-30:
Hello,

I have the same problem. Rolling back to 1.8.10 helped, but it would be nice to use the latest version.

I just started learning electronics with Arduino some weeks ago and it freaked me out first that the upload did not work as expected, even though I did everything as suggested. It took me days to figure out what is the problem (or at least that the IDE restart helps) and to find this ticket with the "temporary" solution.
Just a heads up, that this might makes the learning experience harder for others as well.

Anyways, I'm happy that I found my answers and can go back to build my circuits without being worried :) Great fun so far, thank you for making it possible.