arduino/Arduino #10355: Error: junk at end of line, first unrecognized character valued 0x5

gah4, 2020-06-12:
I just installed the OS X IDE software.

Everything I compile gets the message:

/private/var/folders/2t/xfr4x9715550wm7gv2jbtccm0000h0/T/AppTranslocation/64B411C8-EDCB-4024-81D8-A91F4A60E34C/d/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/._wiring_pulse.S:1: Error: junk at end of line, first unrecognized character valued 0x5

which seems to be in a library routine, not in my code.

This even happens compiling the built-in programs.

Is the OS X version so unused that no-one else found this?

thanks.

facchinm, 2020-06-12:
@gah4 the OSX version of the IDE works well for a lot of users so it might be related to your installation. Can you share which macOS version you are running and the **complete** verbose output from a compile run? You can enable it under Preferences -> Show verbose output during compilation

gah4, 2020-06-12:
I am using OS X 10.12.6

There are many warnings, the one I sent is the error and last.
I might not even have verbose, but these are the messages.

thanks.

/private/var/folders/2t/xfr4x9715550wm7gv2jbtccm0000h0/T/AppTranslocation/64B411C8-EDCB-4024-81D8-A91F4A60E34C/d/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /private/var/folders/2t/xfr4x9715550wm7gv2jbtccm0000h0/T/AppTranslocation/64B411C8-EDCB-4024-81D8-A91F4A60E34C/d/Arduino.app/Contents/Java/hardware -tools /private/var/folders/2t/xfr4x9715550wm7gv2jbtccm0000h0/T/AppTranslocation/64B411C8-EDCB-4024-81D8-A91F4A60E34C/d/Arduino.app/Contents/Java/tools-builder -tools /private/var/folders/2t/xfr4x9715550wm7gv2jbtccm0000h0/T/AppTranslocation/64B411C8-EDCB-4024-81D8-A91F4A60E34C/d/Arduino.app/Contents/Java/hardware/tools/avr -built-in-libraries /private/var/folders/2t/xfr4x9715550wm7gv2jbtccm0000h0/T/AppTranslocation/64B411C8-EDCB-4024-81D8-A91F4A60E34C/d/Arduino.app/Contents/Java/libraries -libraries /Users/gah/Documents/Arduino/libraries -fqbn=arduino:avr:uno -ide-version=10812 -build-path /var/folders/2t/xfr4x9715550wm7gv2jbtccm0000h0/T/arduino_build_252065 -warnings=default -build-cache /var/folders/2t/xfr4x9715550wm7gv2jbtccm0000h0/T/arduino_cache_836921 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=/private/var/folders/2t/xfr4x9715550wm7gv2jbtccm0000h0/T/AppTranslocation/64B411C8-EDCB-4024-81D8-A91F4A60E34C/d/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=/private/var/folders/2t/xfr4x9715550wm7gv2jbtccm0000h0/T/AppTranslocation/64B411C8-EDCB-4024-81D8-A91F4A60E34C/d/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avr-gcc.path=/private/var/folders/2t/xfr4x9715550wm7gv2jbtccm0000h0/T/AppTranslocation/64B411C8-EDCB-4024-81D8-A91F4A60E34C/d/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=/private/var/folders/2t/xfr4x9715550wm7gv2jbtccm0000h0/T/AppTranslocation/64B411C8-EDCB-4024-81D8-A91F4A60E34C/d/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.arduinoOTA.path=/private/var/folders/2t/xfr4x9715550wm7gv2jbtccm0000h0/T/AppTranslocation/64B411C8-EDCB-4024-81D8-A91F4A60E34C/d/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=/private/var/folders/2t/xfr4x9715550wm7gv2jbtccm0000h0/T/AppTranslocation/64B411C8-EDCB-4024-81D8-A91F4A60E34C/d/Arduino.app/Contents/Java/hardware/tools/avr -verbose /var/folders/2t/xfr4x9715550wm7gv2jbtccm0000h0/T/untitled144087628.tmp/sketch_jun11c/sketch_jun11c.ino
/private/var/folders/2t/xfr4x9715550wm7gv2jbtccm0000h0/T/AppTranslocation/64B411C8-EDCB-4024-81D8-A91F4A60E34C/d/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /private/var/folders/2t/xfr4x9715550wm7gv2jbtccm0000h0/T/AppTranslocation/64B411C8-EDCB-4024-81D8-A91F4A60E34C/d/Arduino.app/Contents/Java/hardware -tools /private/var/folders/2t/xfr4x9715550wm7gv2jbtccm0000h0/T/AppTranslocation/64B411C8-EDCB-4024-81D8-A91F4A60E34C/d/Arduino.app/Contents/Java/tools-builder -tools /private/var/folders/2t/xfr4x9715550wm7gv2jbtccm0000h0/T/AppTranslocation/64B411C8-EDCB-4024-81D8-A91F4A60E34C/d/Arduino.app/Contents/Java/hardware/tools/avr -built-in-libraries /private/var/folders/2t/xfr4x9715550wm7gv2jbtccm0000h0/T/AppTranslocation/64B411C8-EDCB-4024-81D8-A91F4A60E34C/d/Arduino.app/Contents/Java/libraries -libraries /Users/gah/Documents/Arduino/libraries -fqbn=arduino:avr:uno -ide-version=10812 -build-path /var/folders/2t/xfr4x9715550wm7gv2jbtccm0000h0/T/arduino_build_252065 -warnings=default -build-cache /var/folders/2t/xfr4x9715550wm7gv2jbtccm0000h0/T/arduino_cache_836921 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=/private/var/folders/2t/xfr4x9715550wm7gv2jbtccm0000h0/T/AppTranslocation/64B411C8-EDCB-4024-81D8-A91F4A60E34C/d/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=/private/var/folders/2t/xfr4x9715550wm7gv2jbtccm0000h0/T/AppTranslocation/64B411C8-EDCB-4024-81D8-A91F4A60E34C/d/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avr-gcc.path=/private/var/folders/2t/xfr4x9715550wm7gv2jbtccm0000h0/T/AppTranslocation/64B411C8-EDCB-4024-81D8-A91F4A60E34C/d/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=/private/var/folders/2t/xfr4x9715550wm7gv2jbtccm0000h0/T/AppTranslocation/64B411C8-EDCB-4024-81D8-A91F4A60E34C/d/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.arduinoOTA.path=/private/var/folders/2t/xfr4x9715550wm7gv2jbtccm0000h0/T/AppTranslocation/64B411C8-EDCB-4024-81D8-A91F4A60E34C/d/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=/private/var/folders/2t/xfr4x9715550wm7gv2jbtccm0000h0/T/AppTranslocation/64B411C8-EDCB-4024-81D8-A91F4A60E34C/d/Arduino.app/Contents/Java/hardware/tools/avr -verbose /var/folders/2t/xfr4x9715550wm7gv2jbtccm0000h0/T/untitled144087628.tmp/sketch_jun11c/sketch_jun11c.ino
Using board 'uno' from platform in folder: /private/var/folders/2t/xfr4x9715550wm7gv2jbtccm0000h0/T/AppTranslocation/64B411C8-EDCB-4024-81D8-A91F4A60E34C/d/Arduino.app/Contents/Java/hardware/arduino/avr
Using core 'arduino' from platform in folder: /private/var/folders/2t/xfr4x9715550wm7gv2jbtccm0000h0/T/AppTranslocation/64B411C8-EDCB-4024-81D8-A91F4A60E34C/d/Arduino.app/Contents/Java/hardware/arduino/avr
[... cut ...]
/private/var/folders/2t/xfr4x9715550wm7gv2jbtccm0000h0/T/AppTranslocation/64B411C8-EDCB-4024-81D8-A91F4A60E34C/d/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/._wiring_pulse.S: Assembler messages:
/private/var/folders/2t/xfr4x9715550wm7gv2jbtccm0000h0/T/AppTranslocation/64B411C8-EDCB-4024-81D8-A91F4A60E34C/d/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/._wiring_pulse.S:1: Error: junk at end of line, first unrecognized character valued 0x5
exit status 1
Error compiling for board Arduino Uno.

facchinm, 2020-06-12:
This line is very suspicious: `Mac OS X 2 � � ATTR � � � � : � : com.apple.quarantine 0081;5ee1fcc3;Firefox;08AF6899-3E92-4574-849E-2DB3A3702C37 This resource fork intentionally left blank ��` 
`._wiring_pulse.S` file doesn't exist while `wiring_pulse.S` does, so I think the mac quarantine could be creating it. Try removing the file from the quarantine using some command like this https://superuser.com/a/28394

gah4, 2020-06-12:
I didn't know about mac quarantine, but it does tell me each time
that I am running it the first time.  I wasn't worried too much about that.
I only recently put Norton on this Mac, as Comcast supplies it free. 
Maybe that is the one that did it.

I will see if I can get it turned off.

thanks,

-- glen

gah4, 2020-06-12:
Thanks much.  I used find to remove all such files, and changed
the attributes so hopefully it won't make any more.  It runs fine now.

I was a little surprised that it doesn't come up with Google search,
that no-one else had this before? 

It is funny, with quarantine discussions so common now, I didn't think
about that, and didn't know about the Apple version at all.

thanks,

-- glen

facchinm, 2020-06-13:
Super, glad it helped. Closing as fixed

closed 2020-06-13 by facchinm

ndreisg, 2020-06-23:
fyi: I had the same issue when I tried to move the Arduino.app to a location outside of /Applications
I created a symlink in /Applications but it didn't help.
Once I moved it back everything worked fine again.
As my SSD is very small I would love to have the opportunity to save my apps on an external drive, maybe you guys can fix this.

gah4, 2020-06-23:
The links in the answer to my question solved it.

You have to delete files with name, I believe, starting with ._
that is, (usually) using find.