arduino/Arduino #10369: [Library Manager] Please add the SparkFun_ADS122C04_ADC_Arduino_Library

PaulZC, 2020-06-15:
Please add the SparkFun_ADS122C04_ADC_Arduino_Library to the library manager:
https://github.com/sparkfun/SparkFun_ADS122C04_ADC_Arduino_Library
Sincere thanks - as always!
Paul

cmaglie, 2020-06-22:
@PaulZC  the system tracked the library as ADS122C0**3** (with the `3` as last digit) as it is written in library.properties is this correct?

PaulZC, 2020-06-22:
Hi Cristian (@cmaglie ),

I'm really sorry. The 3 was just a simple typo. It should have been a 4. Thank you for identifying it!

I have deleted release v1.0.0 and its tag, and have re-released v1.0.0 with the correct name in the library.properties. But I see that the library manager has already picked up the library (with the wrong name). Do I need to do anything else? Will this be corrected automatically? If this needs manual intervention from you then please accept my apologies for the extra work.

(After 30+ years of engineering, I'm still finding new ways to mess things up!)

Very best wishes,

Paul

cmaglie, 2020-06-30:
No worries, we all make mistakes and this one is easily fixed :-)

closed 2020-06-30 by cmaglie

PaulZC, 2020-06-30:
You're a star Cristian (@cmaglie) - thank you,
Paul