arduino/Arduino #10373: Error compiling for board MKR1000

hukufelu, 2020-06-16:
Dear all,

I have been using MKR1000 for some years. Recently, it fails to compile codes even I run blank sketch. However, when I run the same code in UNO, it works. the error message is as follows:

In file included from C:\Users\Joe\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.6\cores\arduino/Arduino.h:24:0,

                 from sketch\sketch_jun16a.ino.cpp:1:

c:\users\joe\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4\lib\gcc\arm-none-eabi\7.2.1\include\stdint.h:9:16: fatal error: stdint.h: No such file or directory

 # include_next <stdint.h>

                ^~~~~~~~~~

compilation terminated.

exit status 1
Error compiling for board Arduino MKR1000.

May I know what is wrong?

facchinm, 2020-06-22:
Hi @hukufelu ,
probably the toolchain extraction went wrong and didn't really copy all the files (or some were quarantined by an antivirus). Try reinstalling the `samd` core with the antivirus off and check if there's enough space in the disk. Let me know if it solves the issue :wink:

per1234, 2020-08-04:
Hi @hukufelu did you ever get the chance to try what facchinm suggested? If so, did it fix the problem?

WebSamuraiDaisuke, 2020-11-30:
Hi, I have a similar issue.

```Error compiling for board M5Stick-C.```
```Error compiling for board M5Stack-Paper```

What should I do?