arduino/Arduino #10374: Problems with Arduino application.

EvgShl, 2020-06-16:
Hello. 
I have the problem with starting Zipped Arduino program.

I wanted to download one of old versions arduino-1.6.5-r5-windows on my windows 10 Home edition x64. OK I unzipped folder. So when I open arduino.exe application in unzipped folder and after I get error message from Arduino, which says:

This application was configured to use a bundled Java Runtime Environment but the runtime is missing or corrupted.

After I press OK button, message goes away, but app does not appear at all.

I thought I don't have java on my new PC. However, the problem is still present.
OK. I deleted zipped version and tried to install via Windows Installer. However, it has NSIS Error which says : Error launching installer. So I had to delete installer file too.

I tried all the same with version ARDUINO 1.8.12 and the same steps had failed too. so I got to delete folders and installer file too again.

I even tried to bring version of arduino-1.6.5-r5-windows zipped version from my laptop to PC on usb stick. Same error. So unfortunately got to delete zipped and unzipped folders.

After that I went to Microsoft Store. I found it, downloaded. The version is ARDUINO 1.8.12. It opens and shows all functions that should be. However, I got a problem with adding up zipped library which I downloaded from www.OSEPP.com official website.

 I wrote about my problem yesterday in Arduino forum, but no answer yet.

Please help.  Could anyone tell which Windows version is OK and which version of Java is good to work with Arduino application?

Best regards!

[image]
[image]
[image]

EvgShl, 2020-06-16:
If there any concerns or questions, please don't hesitate to write.
I need help.
Best Regards!

mshiravi, 2020-06-17:
Halo, If your problem is resolved in the future, please let me know. I have also this problem.
mohammadreza.shiravi@gmail.com

EvgShl, 2020-06-17:
Sure, I will.

GmGniap, 2020-07-23:
if the error is still happening , recommend to test also in [Arduino Web IDE](https://create.arduino.cc/editor/). When using web IDE , if the port is missing , you should follow this [guide](https://create.arduino.cc/getting-started/plugin/welcome?page=2) to install Arduino create plugin. If you also uploaded your zipped library to the Web IDE and it is still showing error , I think it might be your zip package error.

simixd4, 2020-12-31:
Hi!
I had the same problem.
Did you tried debug version?

luongminhduy, 2021-11-14:
You can put your arduino file to C: and solve this problem!

VaibhavPo, 2022-08-23:
The problem is still there with me even after copying files into C:

VaibhavPo, 2022-08-23:
For me it just worked when I again rezipped the files directly in C:/program folder

DasTurn, 2023-04-23:
Install the arduino exe file and the error will go away (it helped me)