arduino/Arduino #10376: Better clean edruino temp folder

BOBILLIER, 2020-06-16:
Today i encountered one compilation problem. I work on one big project with few .ccp and .h, and during my stuff, i have decided to change the name of one my .ccp and .h. I have modified all the calls to those files and update them with new names in other parts of program. After  fine controls, i try to compile, but like i say before, obtain one problem. The error message made reference to new and old name.
After few minutes of recherche, i found solution by deleting completely edruino\temp folder who have conserved old name references.
I think it will be better to completely delete this folder before new compilation to avoid this kind of problems. Best regards
Eric    (Arduino  compiler version 1.8.10)

per1234, 2020-08-01:
>Arduino compiler version 1.8.10

You're using a version of the Arduino IDE that's nearly a year out of date. Please always test with the hourly build before reporting a bug to make sure the issue hasn't already been fixed.

So please download the hourly build of the Arduino IDE and try again:
https://www.arduino.cc/en/Main/Software#hourly
then report back here as to whether the issue still occurs.

per1234, 2020-09-25:
Hi @BOBILLIER. Did you ever get a chance to try this out with the hourly build of the Arduino IDE? If so, does the problem still occur?

BOBILLIER, 2020-09-28:
Hi @per1234 
Not really. Like I said before, this problem arrived when I rename one file in my project. And I have not done that since my request.I think it's not very hard to reproduce.

per1234, 2020-09-30:
OK, it does still occur, but we have an existing issue report (https://github.com/arduino/Arduino/issues/5055) for this bug, so I'm gotin to close this one as a duplicate. Thanks for taking time to make the report!

closed 2020-09-30 by per1234