arduino/Arduino #10379: avrdude: error reading system wide configuration file /avrdude.conf

minhEPFL, 2020-06-17:
Hello,

I have been trying to upload a sketch to my ATMega328P using a programmer, but I get the following problem:

[image]

This problem appeared after a library update from Arduino IDE. Can somebody please help me?

Best regards,

Minh

facchinm, 2020-06-17:
Hi @minhEPFL , it looks something went wrong while extracting `avrdude` package. Try reinstalling avr core 1.8.3 (and make sure no antivirus is kicking in and that there's enough space on disk)

minhEPFL, 2020-06-17:
Hi @facchinm , thank you for the quick reply. I have tried to re-install Arduino IDE 1.8.12, but without success: the same problem is still here. Is that what you meant? If no, can you please explain how to re-install avr core 1.8.3?

facchinm, 2020-06-17:
You need to open Board Manager (from Tools/Boards) and search for AVR, removing core 1.8.3 and then reinstalling it

minhEPFL, 2020-06-17:
Oh my God, thanks a lot @facchinm . Everything works again!

closed 2020-06-17 by minhEPFL

Arun18IT121, 2021-05-17:
avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\ELCOT\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf"

         Using Port                    : COM3
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200

Xasaias, 2021-08-16:
> Hola @minhEPFL , parece que algo salió mal al extraer el `avrdude`paquete. Intente reinstalar avr core 1.8.3 (y asegúrese de que no esté funcionando ningún antivirus y de que haya suficiente espacio en el disco)

Como se hace eso me esta pasando pero no puedo arreglarlo ayuda por favor