arduino/Arduino #10390: [Library manager] Please add MCP9808_RT library

RobTillaart, 2020-06-19:
Please add https://github.com/RobTillaart/MCP9808_RT
thanks

per1234, 2020-06-19:
Hi @RobTillaart. Each library added to the Library Manager index must have a unique library.properties `name` value (regardless of case). The `name` value used in your library has already been taken:
https://github.com/JChristensen/MCP9808

If you would still like your library added to the index, you'll need to change the `name` value to a unique name, then either update the `version` value in library.properties and make a new release/tag or else redo the existing tag.

After that is done, please comment here to let us know we can proceed. Thanks!

RobTillaart, 2020-06-20:
@per1234 
Changed name to MCP9808_RT

cmaglie, 2020-06-22:
Done.

closed 2020-06-22 by cmaglie