arduino/Arduino #10402: Hourly build  16. June - OS X don't work

stoneblock, 2020-06-22:
Hi,

I just load (two-times) the build from 16.June and if I start this version I always get:

"Arduino" is damaged and cannot be opened. It is recommended to move the object to the trash. (OS X 10.12.6)
 
Thx

facchinm, 2020-06-22:
Hi @stoneblock ,
does official IDE 1.8.13 have the same problem?

stoneblock, 2020-06-22:
I've downloaded it (I'm in Hollyday and a small amount of data ... ) and with this it's no problem.
for me fine.
I think as this both versions on the same day there are no much differences

Thanks

stoneblock, 2020-06-23:
> Hi @stoneblock ,
> does official IDE 1.8.13 have the same problem?

i had "answered" - but only in the direct not to you
yes tested and no problems with it
thx

cmaglie, 2020-06-24:
The hourly build has been fixed too.

closed 2020-06-24 by cmaglie