arduino/Arduino #10404: [Library Manager] Please add the EASE_ArduinoCode Library

esmacat-hb, 2020-06-22:
Please add the EASE_ArduinoCode Library : https://bitbucket.org/harmonicbionics/ease_arduinocode/src/

Thanks,
Esmacat

per1234, 2020-06-22:
Hi @esmacat-hb. You need to create a [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

esmacat-hb, 2020-06-22:
Hi @per1234. A tag with the appropriate version number has been added to the repository. 

Thanks,
Esmacat.

cmaglie, 2020-06-30:
Done.

closed 2020-06-30 by cmaglie

esmacat-hb, 2021-05-03:
Hello! We made an update to our Arduino library (v0.3.0). Please add the update to the library manager whenever possible.

Thanks,
Esmacat