arduino/Arduino #10409: Cannot use two boards simultaneously

jzeggelaar, 2020-06-23:
With the 1.9 beta the board and port settings appear to be shared between IDE instances. So I can no longer simultaneously edit and debug two sketches for two different boards. I use this for a project where two boards communicate (using 12c).  I'm pretty sure this used to work with 1.8, but I cannot test this because 1.8 suddenly no longer works for me on Windows 10 (caused by Microsoft update?)

dotto59, 2020-07-03:
Open the first project for the first board, then **don't** open the second by using the menu (File, Open) but instead launch another IDE instance either from the Start menu, or by right-clicking the *.ino file of the second project and select "Open with.." "Arduino".
I think that when you use File-Open the instance/thread is the same (thus sharing everything), but if you run two distinct executables instances you can do what you need. I discovered this for a distributed project I'm working on, requiring three different boards (Arduino UNO, a WeMos D1 R2 connected to UNO via serial, and a WeMos D1 Mini connected via WiFi to D1 R2).
Give it a try...

jzeggelaar, 2020-07-03:
I just discovered that today. For some reason I always do it like that on my laptop (where I didn’t have the problem) even, as it turned out, after updating Windows 10.

So, sorry for this false positive. And thanks for getting back to me anyway.

Jan

per1234, 2020-07-03:
Thanks to @dotto59 for the assist and thanks also to @jzeggelaar for taking the time to post an update. Since it sounds like this is resolved, I'll go ahead and close.

closed 2020-07-03 by per1234