arduino/Arduino #10416: LibraryManager - please add 'RLEBitmap'

MHotchin, 2020-06-24:
Repository:

https://github.com/MHotchin/RLEBitmap

per1234, 2020-06-24:
Hi @MHotchin. Libraries that contain a .exe file are not permitted to be added to the Arduino Library Manager index. Your library contains such a file:
https://github.com/MHotchin/RLEBitmap/blob/master/extras/Images/bmper.exe

If you still want your library added to the Library Manager index, you'll need to remove the file and then either create a new tag/release or else redo the existing tag. After that, please comment here so we'll know we can proceed. Otherwise, you can close this issue.

More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

MHotchin, 2020-06-24:
Done

per1234, 2020-06-24:
The repository is now compliant, however the Library Manager indexer works from Git tags. At the time of the tag, the repository was still not compliant. For this reason you either need to create a new release/tag (making sure to first update the `version` value in library.properties) or else redo the existing tag.

Once that's done, please comment here.

MHotchin, 2020-06-24:
I had already deleted the original tag, and created it again after deleting the file.

per1234, 2020-06-24:
You must either have not pushed the tag or else didn't point it at the tip of the branch then because the tag still points to the commit from before you removed the .exe file:
https://github.com/MHotchin/RLEBitmap/compare/1.0.0...master

MHotchin, 2020-06-24:
I think it's correct now.

cmaglie, 2020-06-30:
Done.

closed 2020-06-30 by cmaglie